NVIDIA-AI-IOT/Lidar_AI_Solution #1: How can I convert a lidar pcd file to points.tensor ?

lucianzhong, 2023-05-17:
Thanks for this great work. I can run on orin with 45 ms time consumption per frame which is very impressive. 
Now, I want to try with my own data. How can I convert a lidar pcd file to points.tensor ?

hopef, 2023-05-17:
[Here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/51f4641307f2fa01f0955b66d5e209a0a9199bec/CUDA-BEVFusion/tool/dump-data.py#L80) you can see how to store point clouds to tensor file.

hopef, 2023-05-17:
The tensor file is a straightforward format for data exchange in Python and C++.

The primary code for it is [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/tool/tensor.py) and [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/51f4641307f2fa01f0955b66d5e209a0a9199bec/CUDA-BEVFusion/src/common/tensor.cu#L437)

lucianzhong, 2023-05-17:
Get you. Thanks