NVIDIA-AI-IOT/Lidar_AI_Solution #2: Inference results issue

lucianzhong, 2023-05-18:
I have tried my own data with CUDA-BEVFusion and BEVFusion directly without training. The BEVFusion gives good infercence results. But  CUDA-BEVFusion can not detect person,the vehicle detection is good. Any clue for this?
Best regards

mchi-zg, 2023-05-22:
is it possible to share your input data?

lucianzhong, 2023-05-22:
Sure. Please give me your e-mail. I am current on a business trip. I will send you input data tomorrow or the day after tomorrow

lucianzhong, 2023-05-24:
Has sent the data. Please check QQ mail. Thanks a lot

hopef, 2023-05-24:
According to the data you provided. I found that the lidar and camera do not match. This may be missing a transformation.

lucianzhong, 2023-05-25:
Thanks a lot for checking my data. I will try to fix the match problem first.

closed 2023-05-25 by lucianzhong