NVIDIA-AI-IOT/Lidar_AI_Solution #4: plugin for SparseConvolution

shuyuan-wang, 2023-05-18:
Hi:
I can see that there is custom onnx node "SparseConvolution" in Centerpoint middle encoder onnx file. However, I can't find its plugin or how this onnx is converted to tensorrt engine. Can you guide me to where to find it?

hopef, 2023-05-18:
It is not a plugin. Because the data-depend characteristic can not be supported in the current version of TensorRT. So we implemented a tiny engine in a standalone way.

hopef, 2023-05-18:
So it only needs to build the execution graph from the ONNX file, which doesn't need to convert to a TensorRT engine.

shuyuan-wang, 2023-05-18:
Got it, so the spconv part won't be converted to an tensorrt engine?

shuyuan-wang, 2023-05-18:
Got it

shuyuan-wang, 2023-05-18:
Any plan on adding inverse sparse convolution?

hopef, 2023-05-18:
This support will be in the queue.

closed 2023-05-18 by shuyuan-wang