NVIDIA-AI-IOT/Lidar_AI_Solution #14: Version of Tensorrt

Oh-Cest-la-vie, 2023-05-19:
Hello

I see in the README of CUDA-CenterPoint, you use 'CUDA 11.4 + cuDNN 8.4.1 + TensorRT 8.4.12.5' at Tesla Platform, but I can't find the version 8.4.12.5 of Tensorrt. In addition, I see that  you set **CUDA_TOOLKIT_ROOT_DIR=/root/.kiwi/lib/cuda-11.8 and  TENSORRT_ROOT=/root/.kiwi/lib/TensorRT-8.5.3.1-cuda11x** in the CmakeLists of CUDA-CenterPoint. So should I follow the version in the CmakeLists or README?

Thanks

hopef, 2023-05-19:
There is no strict requirement for the TensorRT version in this repo. Just newer is better recommended.

Oh-Cest-la-vie, 2023-05-19:
And how about CUDA, is it limited?

hopef, 2023-05-19:
For the CUDA version, 11.4 or higher is better.

Oh-Cest-la-vie, 2023-05-19:
Got it, thanks