NVIDIA-AI-IOT/Lidar_AI_Solution #15: [libprotobuf ERROR google/protobuf/text_format.cc:298] Error parsing text-format onnx2trt_onnx.ModelProto: 1:9: Message type "onnx2trt_onnx.ModelProto" has no field named "version".

blazeCCC, 2023-05-19:
thanks for your great work! I meet a question when i run` "bash tool/build.trt.sh"`, the error is :
`[libprotobuf ERROR google/protobuf/text_format.cc:298] Error parsing text-format onnx2trt_onnx.ModelProto: 1:9: Message type "onnx2trt_onnx.ModelProto" has no field named "version".`

My environment:
cuda11.7
tensorRT 8.5.3
ubuntu18
protobuf 3.6.1

Whether this project requires a higher version of protobuf？

blazeCCC, 2023-05-19:
Whole log is:
`&&&& RUNNING TensorRT.trtexec [TensorRT v8503] # trtexec --onnx=model/rpn_centerhead_sim.onnx --saveEngine=model/rpn_centerhead_sim.plan.8503 --workspace=4096 --fp16 --outputIOFormats=fp16:chw --inputIOFormats=fp16:chw --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed
[05/19/2023-15:33:07] [W] --workspace flag has been deprecated by --memPoolSize flag.
[05/19/2023-15:33:07] [I] === Model Options ===
[05/19/2023-15:33:07] [I] Format: ONNX
[05/19/2023-15:33:07] [I] Model: model/rpn_centerhead_sim.onnx
[05/19/2023-15:33:07] [I] Output:
[05/19/2023-15:33:07] [I] === Build Options ===
[05/19/2023-15:33:07] [I] Max batch: explicit batch
[05/19/2023-15:33:07] [I] Memory Pools: workspace: 4096 MiB, dlaSRAM: default, dlaLocalDRAM: default, dlaGlobalDRAM: default
[05/19/2023-15:33:07] [I] minTiming: 1
[05/19/2023-15:33:07] [I] avgTiming: 8
[05/19/2023-15:33:07] [I] Precision: FP32+FP16
[05/19/2023-15:33:07] [I] LayerPrecisions: 
[05/19/2023-15:33:07] [I] Calibration: 
[05/19/2023-15:33:07] [I] Refit: Disabled
[05/19/2023-15:33:07] [I] Sparsity: Disabled
[05/19/2023-15:33:07] [I] Safe mode: Disabled
[05/19/2023-15:33:07] [I] DirectIO mode: Disabled
[05/19/2023-15:33:07] [I] Restricted mode: Disabled
[05/19/2023-15:33:07] [I] Build only: Disabled
[05/19/2023-15:33:07] [I] Save engine: model/rpn_centerhead_sim.plan.8503
[05/19/2023-15:33:07] [I] Load engine: 
[05/19/2023-15:33:07] [I] Profiling verbosity: 2
[05/19/2023-15:33:07] [I] Tactic sources: Using default tactic sources
[05/19/2023-15:33:07] [I] timingCacheMode: local
[05/19/2023-15:33:07] [I] timingCacheFile: 
[05/19/2023-15:33:07] [I] Heuristic: Disabled
[05/19/2023-15:33:07] [I] Preview Features: Use default preview flags.
[05/19/2023-15:33:07] [I] Input(s): fp16:chw
[05/19/2023-15:33:07] [I] Output(s): fp16:chw
[05/19/2023-15:33:07] [I] Input build shapes: model
[05/19/2023-15:33:07] [I] Input calibration shapes: model
[05/19/2023-15:33:07] [I] === System Options ===
[05/19/2023-15:33:07] [I] Device: 0
[05/19/2023-15:33:07] [I] DLACore: 
[05/19/2023-15:33:07] [I] Plugins:
[05/19/2023-15:33:07] [I] === Inference Options ===
[05/19/2023-15:33:07] [I] Batch: Explicit
[05/19/2023-15:33:07] [I] Input inference shapes: model
[05/19/2023-15:33:07] [I] Iterations: 10
[05/19/2023-15:33:07] [I] Duration: 3s (+ 200ms warm up)
[05/19/2023-15:33:07] [I] Sleep time: 0ms
[05/19/2023-15:33:07] [I] Idle time: 0ms
[05/19/2023-15:33:07] [I] Streams: 1
[05/19/2023-15:33:07] [I] ExposeDMA: Disabled
[05/19/2023-15:33:07] [I] Data transfers: Enabled
[05/19/2023-15:33:07] [I] Spin-wait: Disabled
[05/19/2023-15:33:07] [I] Multithreading: Disabled
[05/19/2023-15:33:07] [I] CUDA Graph: Disabled
[05/19/2023-15:33:07] [I] Separate profiling: Enabled
[05/19/2023-15:33:07] [I] Time Deserialize: Disabled
[05/19/2023-15:33:07] [I] Time Refit: Disabled
[05/19/2023-15:33:07] [I] NVTX verbosity: 2
[05/19/2023-15:33:07] [I] Persistent Cache Ratio: 0
[05/19/2023-15:33:07] [I] Inputs:
[05/19/2023-15:33:07] [I] === Reporting Options ===
[05/19/2023-15:33:07] [I] Verbose: Enabled
[05/19/2023-15:33:07] [I] Averages: 10 inferences
[05/19/2023-15:33:07] [I] Percentiles: 90,95,99
[05/19/2023-15:33:07] [I] Dump refittable layers:Disabled
[05/19/2023-15:33:07] [I] Dump output: Disabled
[05/19/2023-15:33:07] [I] Profile: Enabled
[05/19/2023-15:33:07] [I] Export timing to JSON file: 
[05/19/2023-15:33:07] [I] Export output to JSON file: 
[05/19/2023-15:33:07] [I] Export profile to JSON file: 
[05/19/2023-15:33:07] [I] 
[05/19/2023-15:33:07] [I] === Device Information ===
[05/19/2023-15:33:07] [I] Selected Device: NVIDIA GeForce RTX 4070 Ti
[05/19/2023-15:33:07] [I] Compute Capability: 8.9
[... 46 lines cut ...]
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::Reorg_TRT version 1
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::ResizeNearest_TRT version 1
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::ROIAlign_TRT version 1
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::RPROI_TRT version 1
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::ScatterND version 1
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::SeqLen2Spatial version 1
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::SpecialSlice_TRT version 1
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::SplitGeLU version 1
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::Split version 1
[05/19/2023-15:33:07] [V] [TRT] Registered plugin creator - ::VoxelGeneratorPlugin version 1
[05/19/2023-15:33:09] [I] [TRT] [MemUsageChange] Init CUDA: CPU +362, GPU +0, now: CPU 370, GPU 2515 (MiB)
[05/19/2023-15:33:09] [V] [TRT] Trying to load shared library libnvinfer_builder_resource.so.8.6.0
[05/19/2023-15:33:09] [V] [TRT] Loaded shared library libnvinfer_builder_resource.so.8.6.0
[05/19/2023-15:33:18] [I] [TRT] [MemUsageChange] Init builder kernel library: CPU +1209, GPU +249, now: CPU 1655, GPU 2719 (MiB)
[05/19/2023-15:33:18] [W] [TRT] CUDA lazy loading is not enabled. Enabling it can significantly reduce device memory usage and speed up TensorRT initialization. See "Lazy Loading" section of CUDA documentation https://docs.nvidia.com/cuda/cuda-c-programming-guide/index.html#lazy-loading
[05/19/2023-15:33:18] [I] Start parsing network model
[... cut ...]
[05/19/2023-15:33:18] [I] Finish parsing network model
[05/19/2023-15:33:18] [E] Parsing model failed
[05/19/2023-15:33:18] [E] Failed to create engine from model or file.
[05/19/2023-15:33:18] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v8503] # trtexec --onnx=model/rpn_centerhead_sim.onnx --saveEngine=model/rpn_centerhead_sim.plan.8503 --workspace=4096 --fp16 --outputIOFormats=fp16:chw --inputIOFormats=fp16:chw --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed
`

hopef, 2023-05-19:
It may be that your model/rpn_centerhead_sim.onnx is not fully downloaded. 
Since onnx is stored using lfs, you will need to install git-lfs and then pull the data.

Or download it directly via the [url](https://media.githubusercontent.com/media/NVIDIA-AI-IOT/Lidar_AI_Solution/master/CUDA-CenterPoint/model/rpn_centerhead_sim.onnx)

blazeCCC, 2023-05-19:
Problem solved. Thank you