NVIDIA-AI-IOT/Lidar_AI_Solution #21: 大佬，编译报错

xqh5201314, 2023-05-22:
make[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:112：CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-vtransform.cu.o] 错误 1
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
CMake Error at bevfusion_core_generated_lidar-voxelization.cu.o.Release.cmake:280 (message):
  Error generating file
  /home/chenyang/Lidar_AI_Solution_2/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/bevfusion/./bevfusion_core_generated_lidar-voxelization.cu.o

mchi-zg, 2023-05-22:
please elaborate your compile environment/setup

xqh5201314, 2023-05-23:
ubuntu22.04 
rtx3090 
cuda 11.6
cudnn 8.6.0
tensorrt 8.5.2

hopef, 2023-05-23:
It seems to be an error caused by compiling with c++11. But this repo requires c++14 or higher.
Also, you should check if gcc version is too low.

xqh5201314, 2023-05-23:
gcc version is the lastest.