NVIDIA-AI-IOT/Lidar_AI_Solution #25: Segmentation fault (core dumped) when run tool/run.sh

Mandylove1993, 2023-05-23:
I follow the readme,prepare the models and data,build engine,,but when run tool/run.sh,the follow error exist:
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 	0.44275 ms
[⏰ [NoSt] ImageNrom]: 	5.44451 ms
[⏰ Lidar Backbone]: 	3.04624 ms
[⏰ Camera Depth]: 	0.03789 ms
[⏰ Camera Backbone]: 	2.55898 ms
[⏰ Camera Bevpool]: 	0.35421 ms
[⏰ VTransform]: 	0.61219 ms
[⏰ Transfusion]: 	1.63635 ms
[⏰ Head BoundingBox]: 	3.45373 ms
Total: 11.700 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 	0.44605 ms
[⏰ [NoSt] ImageNrom]: 	5.41389 ms
[⏰ Lidar Backbone]: 	3.03587 ms
[⏰ Camera Depth]: 	0.03789 ms
[⏰ Camera Backbone]: 	2.55693 ms
[⏰ Camera Bevpool]: 	0.36045 ms
[⏰ VTransform]: 	0.60826 ms
[⏰ Transfusion]: 	1.62582 ms
[⏰ Head BoundingBox]: 	3.45293 ms
Total: 11.678 ms
=============================================
tool/run.sh: line 41:  2643 Segmentation fault      (core dumped) ./build/bevfusion $DEBUG_DATA $DEBUG_MODEL $DEBUG_PRECISION

Thank for your reply!

kunnalparihar, 2023-05-23:
Same happens in my system. After running the inference, code did not exit correctly. It just throw a segmentation fault.

hopef, 2023-05-23:
It seems to happen in the visualization function, and I need to debug this issue.

hopef, 2023-05-29:
Same as #35

closed 2023-05-29 by hopef