NVIDIA-AI-IOT/Lidar_AI_Solution #28: Generate tiny engine for spconv

kunnalparihar, 2023-05-23:
Thanks for the great work. I already tested inference on example data. In issue #4 , you mentioned that you generated a tiny engine in a standalone way for spconv. Can you elaborate on how can I generate tiny engine if I make some changes to spconv code in cuda-bevfusion ?

hopef, 2023-05-23:
Since the tiny engine is loaded and executed via ONNX, you only need to use the code [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/export-scn.py) to export your modified ONNX.

hopef, 2023-05-23:
This script will [trace](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/8b71cb006d434b4c20317c66121da59b99b4508e/CUDA-BEVFusion/qat/lean/exptool.py#L396) your model and generate an ONNX file.

For your modification, you should refer to the notes [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution#note).

Thanks

hopef, 2023-05-23:
If you would like more support, feel free to post a comment.

Thanks.

closed 2023-05-29 by hopef

kunnalparihar, 2023-05-29:
@hopef Thanks for your support