NVIDIA-AI-IOT/Lidar_AI_Solution #34: centerpoint engine setup failed

bigsheep2018, 2023-05-24:
hello Lidar_AI_Solution team @hopef ,

type bash tool/build.trt.sh, error occurs as follows:
[05/24/2023-16:26:52] [E] Error[10]: [optimizer.cpp::computeCosts::1855] Error Code 10: Internal Error (Could not find any implementation for node ConvTranspose_56 + BatchNormalization_57 + Relu_58.)
[05/24/2023-16:26:52] [E] Error[2]: [builder.cpp::buildSerializedNetwork::417] Error Code 2: Internal Error (Assertion enginePtr != nullptr failed.)
[05/24/2023-16:26:52] [E] Engine could not be created from network
[05/24/2023-16:26:52] [E] Building engine failed
[05/24/2023-16:26:52] [E] Failed to create engine from model or file.
[05/24/2023-16:26:52] [E] Engine set up failed
[rpn_centerhead_sim.8503.log](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/files/11552433/rpn_centerhead_sim.8503.log)

My environment is 
CUDA 11.4
TensorRT 8.5.3.1
GPU: 3060
Ubuntu 20.04

The full log is attached as well.
It seems the error comes from the onnx file. 
Please help figure it out.
Thanks in advance.

hopef, 2023-05-25:
You need to check the correctness of your tensorRT configuration. Because from the log file you gave, the TensorRT version is v8501. But your comment is 8.5.3.1.

bigsheep2018, 2023-05-25:
hello @hopef ,

I modified the path and get the correct 8503 version trtexec (corresponding TensorRT version is 8.5.3.1) as the CMakeLists.txt mentioned [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-CenterPoint/CMakeLists.txt)
if(${arch}     STREQUAL  "aarch64")
  # set(TENSORRT_ROOT         /root/.kiwi/lib/TensorRT-8.5.3.1-cuda11x)
  # set(CUDA_TOOLKIT_ROOT_DIR /root/.kiwi/lib/cuda-11.8)
elseif(${arch} STREQUAL "x86_64")
  set(TENSORRT_ROOT         /root/.kiwi/lib/TensorRT-8.5.3.1-cuda11x)
  set(CUDA_TOOLKIT_ROOT_DIR /root/.kiwi/lib/cuda-11.8)
endif()

The error still exist as follows 
[05/25/2023-09:32:47] [E] Error[10]: [optimizer.cpp::computeCosts::1855] Error Code 10: Internal Error (Could not find any implementation for node ConvTranspose_56 + BatchNormalization_57 + Relu_58.)
[05/25/2023-09:32:47] [E] Error[2]: [builder.cpp::buildSerializedNetwork::417] Error Code 2: Internal Error (Assertion enginePtr != nullptr failed.)
[05/25/2023-09:32:47] [E] Engine could not be created from network
[05/25/2023-09:32:47] [E] Building engine failed
[05/25/2023-09:32:47] [E] Failed to create engine from model or file.
[05/25/2023-09:32:47] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v8503] # trtexec --onnx=model/rpn_centerhead_sim.onnx --saveEngine=model/rpn_centerhead_sim.plan.8503 --workspace=4096 --fp16 --outputIOFormats=fp16:chw --inputIOFormats=fp16:chw --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed

The full log is attached: 
[rpn_centerhead_sim.8503.log](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/files/11560383/rpn_centerhead_sim.8503.log)

If the tensorRT version(8503 && 8.5.3.1) is not correct, please provide the version # or a link to download it.

bigsheep2018, 2023-05-25:
> Maybe you could try to change **trtexec** to **YOUR_PATH_TENSORRT/bin/trtexec** in build.trt.sh. @bigsheep2018

Modified build.trt.sh as follows
trt_version=8503

if [ ! -f "model/rpn_centerhead_sim.plan.${trt_version}" ]; then
    echo Building the model: model/rpn_centerhead_sim.plan.${trt_version}, this will take 2 minutes. Wait a moment 🤗🤗🤗~.
    /home/lin/workspace/TensorRT-8.5.3.1/bin/trtexec --onnx=model/rpn_centerhead_sim_export.onnx \
        --saveEngine=model/rpn_centerhead_sim.plan.${trt_version} \
        --workspace=4096 --fp16 --outputIOFormats=fp16:chw \
        --inputIOFormats=fp16:chw --verbose --dumpLayerInfo \
        --dumpProfile --separateProfileRun \
        --profilingVerbosity=detailed > model/rpn_centerhead_sim.${trt_version}.log 2>&1

    rm -rf model/rpn_centerhead_sim.plan
    dir=`pwd`
    ln -s ${dir}/model/rpn_centerhead_sim.plan.${trt_version} model/rpn_centerhead_sim.plan
else
    echo Model model/rpn_centerhead_sim.plan.${trt_version} already build 🙋🙋🙋.
fi

and got same error: 
[05/25/2023-10:07:53] [E] Error[10]: [optimizer.cpp::computeCosts::1855] Error Code 10: Internal Error (Could not find any implementation for node ConvTranspose_56 + BatchNormalization_57 + Relu_58.)
[05/25/2023-10:07:53] [E] Error[2]: [builder.cpp::buildSerializedNetwork::417] Error Code 2: Internal Error (Assertion enginePtr != nullptr failed.)
[05/25/2023-10:07:53] [E] Engine could not be created from network
[05/25/2023-10:07:53] [E] Building engine failed
[05/25/2023-10:07:53] [E] Failed to create engine from model or file.
[05/25/2023-10:07:53] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v8503] # /home/lin/workspace/TensorRT-8.5.3.1/bin/trtexec --onnx=model/rpn_centerhead_sim.onnx --saveEngine=model/rpn_centerhead_sim.plan.8503 --workspace=4096 --fp16 --outputIOFormats=fp16:chw --inputIOFormats=fp16:chw --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed

full log: 
[rpn_centerhead_sim.8503.log](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/files/11560501/rpn_centerhead_sim.8503.log)

bigsheep2018, 2023-05-25:
@hopef  @Oh-Cest-la-vie
I also tried exporting onnx (named as rpn_centerhead_sim_export.onnx) from original epoch_20.pth using [tool/export_neck_head.py](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-CenterPoint/tool/export_neck_head.py) and got similar error:
```
[05/25/2023-10:19:15] [E] Error[10]: [optimizer.cpp::computeCosts::1855] Error Code 10: Internal Error (Could not find any implementation for node /neck/deblocks.1/deblocks.1.0/ConvTranspose + /neck/deblocks.1/deblocks.1.1/BatchNormalization + /neck/deblocks.1/deblocks.1.2/Relu.)
[05/25/2023-10:19:15] [E] Error[2]: [builder.cpp::buildSerializedNetwork::417] Error Code 2: Internal Error (Assertion enginePtr != nullptr failed.)
[05/25/2023-10:19:15] [E] Engine could not be created from network
[05/25/2023-10:19:15] [E] Building engine failed
[05/25/2023-10:19:15] [E] Failed to create engine from model or file.
[05/25/2023-10:19:15] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v8503] # /home/lin/workspace/TensorRT-8.5.3.1/bin/trtexec --onnx=model/rpn_centerhead_sim_export.onnx --saveEngine=model/rpn_centerhead_sim.plan.8503 --workspace=4096 --fp16 --outputIOFormats=fp16:chw --inputIOFormats=fp16:chw --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed
```
full log:
[rpn_centerhead_sim.8503.log](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/files/11560554/rpn_centerhead_sim.8503.log)

I will try to read onnx with tensorRT c++ api instead. 
Please let me know, if there is any progress of this issue.

shouxieai, 2023-05-25:
It would be best to consider adding your TensorRT/lib path to LD_LIBRARY_PATH.

hopef, 2023-05-25:
It would be best to consider adding your TensorRT/lib path to LD_LIBRARY_PATH.

bigsheep2018, 2023-05-25:
@hopef  I do not think it is a path problem as I can type command "trtexec" directly without export any path. 
The ~/.bashrc has been modified as follows

```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda/lib64
export PATH=$PATH:/usr/local/cuda/bin
export CUDA_HOME=/usr/local/cuda

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/lin/workspace/TensorRT-8.5.3.1/lib
export PATH=$PATH:/home/lin/workspace/TensorRT-8.5.3.1/bin
```

bigsheep2018, 2023-05-25:
update @hopef @Oh-Cest-la-vie 
After re-cloning the repo and restart the pc, everything seems fine now

GPU 3060 performance: 
Perf Report: 
    Voxelization: 1.82443 ms.
    3D Backbone: 13.4205 ms.
    RPN + Head: 8.78339 ms.
    Decode + NMS: 3.23226 ms.
    Total: 27.2606 ms.

Reboot is always a surprising  solution.

closed 2023-05-25 by bigsheep2018