NVIDIA-AI-IOT/Lidar_AI_Solution #35: BEVFUSION, PTX JIT compilation failed, code = cudaErrorInvalidPtx

uefall, 2023-05-25:
root@fe3435b23fa0:/mnt/Lidar_AI_Solution-master/CUDA-BEVFusion# ./tool/run.sh 
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: OFF
||
||  TensorRT: /usr/lib/x86_64-linux-gnu/
||  CUDA: /usr/local/cuda
||  CUDNN: /usr/lib/x86_64-linux-gnu/
==========================================================
Try to get the current device SM
Current CUDA SM: 80
Configuration done!
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/Lidar_AI_Solution-master/CUDA-BEVFusion/build
Consolidate compiler generated dependencies of target bevfusion_core
[ 63%] Built target bevfusion_core
Consolidate compiler generated dependencies of target bevfusion
[ 68%] Building CXX object CMakeFiles/bevfusion.dir/src/main.cpp.o
[ 72%] Linking CXX executable bevfusion
[100%] Built target bevfusion
Create by resnet50int8, int8
CUDA Runtime error create_frustum_kernel # a PTX JIT compilation failed, code = cudaErrorInvalidPtx [ 218 ] in file /mnt/Lidar_AI_Solution-master/CUDA-BEVFusion/src/bevfusion/camera-geometry.cu:209
./tool/run.sh: line 41:   583 Aborted                 (core dumped) ./build/bevfusion $DEBUG_DATA $DEBUG_MODEL $DEBUG_PRECISION

how to solve this?

uefall, 2023-05-25:
可能是我cuda和driver版本不匹配，我会尝试下更新driver

uefall, 2023-05-25:
更新driver后能跑了，但是出现了和 #25 一样的错误，崩在了visualize里

uefall, 2023-05-25:
定位到cuosd_draw_text有问题，注释掉这一行正常运行。
issue close

closed 2023-05-25 by uefall

hopef, 2023-05-29:
我了解到，可能是你的字体文件并没有通过git-lfs下载。导致字体加载失败出现的错误。