NVIDIA-AI-IOT/Lidar_AI_Solution #37: export onnx error

areyouok12, 2023-05-25:
感谢作者提供这么优秀的rpo，我在导出自己训练的BEVfusion模型在运行export-scn.py导出的lidar.backbone.xyz.onnx如图所示：
[image]

运行export-scn.py脚本控制台打印：
[image]

hopef, 2023-05-25:
You can refer to [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/8b71cb006d434b4c20317c66121da59b99b4508e/CUDA-BEVFusion/qat/ptq.py#L145).
It handled the fusion about relu. Because libspconv.so requires relu to be added to the attributes of the sparse convolution.

areyouok12, 2023-05-25:
So how should I modify the code for export-scn.py

hopef, 2023-05-25:
There should not be a relu operation after the conv of the model when exporting onnx. 
It should be merged into the attributes of the conv.

closed 2023-05-29 by hopef

liuxingwt, 2023-06-14:
> So how should I modify the code for export-scn.py

Hello, I met the same question with you. Have you solved it?

YueSun0609, 2023-09-27:
> 感谢作者提供这么优秀的rpo，我在导出自己训练的BEVfusion模型在运行export-scn.py导出的lidar.backbone.xyz.onnx如图所示： [image]
> 
> 运行export-scn.py脚本控制台打印： [image]

Hi, did you open the weights of the onnx model in netron? I found the weights' data type is still float 32.

sangjinpark97, 2023-10-17:
@hopef, Hi. Here is my results of tracing lidar-scn.

```shell
   --> SparseConvolutionQunat0[subm] -> Input 0, Output 1
   --> SparseConvolutionQunat1[subm] -> Input 1, Output 2
   --> SparseConvolutionQunat2[subm] -> Input 2, Output 3
   --> QuantAdd0 -> Input 3 + 1, Output 4
   --> ReLU0 -> Input 4, Output 5
   --> SparseConvolutionQunat3[subm] -> Input 5, Output 6
   --> SparseConvolutionQunat4[subm] -> Input 6, Output 7
   --> QuantAdd1 -> Input 7 + 5, Output 8
   --> ReLU1 -> Input 8, Output 9
   --> SparseConvolutionQunat5[conv] -> Input 9, Output 10
   --> SparseConvolutionQunat6[subm] -> Input 10, Output 11
   --> SparseConvolutionQunat7[subm] -> Input 11, Output 12
   --> QuantAdd2 -> Input 12 + 10, Output 13
   --> ReLU2 -> Input 13, Output 14
   --> SparseConvolutionQunat8[subm] -> Input 14, Output 15
   --> SparseConvolutionQunat9[subm] -> Input 15, Output 16
   --> QuantAdd3 -> Input 16 + 14, Output 17
   --> ReLU3 -> Input 17, Output 18
   --> SparseConvolutionQunat10[conv] -> Input 18, Output 19
   --> SparseConvolutionQunat11[subm] -> Input 19, Output 20
   --> SparseConvolutionQunat12[subm] -> Input 20, Output 21
   --> QuantAdd4 -> Input 21 + 19, Output 22
   --> ReLU4 -> Input 22, Output 23
   --> SparseConvolutionQunat13[subm] -> Input 23, Output 24
   --> SparseConvolutionQunat14[subm] -> Input 24, Output 25
   --> QuantAdd5 -> Input 25 + 23, Output 26
   --> ReLU5 -> Input 26, Output 27
   --> SparseConvolutionQunat15[conv] -> Input 27, Output 28
   --> SparseConvolutionQunat16[subm] -> Input 28, Output 29
   --> SparseConvolutionQunat17[subm] -> Input 29, Output 30
   --> QuantAdd6 -> Input 30 + 28, Output 31
   --> ReLU6 -> Input 31, Output 32
   --> SparseConvolutionQunat18[subm] -> Input 32, Output 33
   --> SparseConvolutionQunat19[subm] -> Input 33, Output 34
   --> QuantAdd7 -> Input 34 + 32, Output 35
   --> ReLU7 -> Input 35, Output 36
   --> SparseConvolutionQunat20[conv] -> Input 36, Output 37
   --> ToDense0[[180, 180, 2]][[1, 128, 180, 180, 2]] -> Input 37, Output 38
   --> Permute0[(0, 1, 4, 2, 3)][[1, 128, 2, 180, 180]] -> Input 38, Output 39
   --> Reshape0[(1, 256, 180, 180)] -> Input 39, Output 40
```

Does it seem to be correct?