NVIDIA-AI-IOT/Lidar_AI_Solution #39: 执行bash tool/run.sh报错

HZQNPU, 2023-05-26:
在执行bash tool/run.sh的时候报错，请问该怎么解决？
fatal error: stb_image.h: 没有那个文件或目录
 #include <stb_image.h>
          ^~~~~~~~~~~~~
compilation terminated.
CMakeFiles/bevfusion.dir/build.make:504: recipe for target 'CMakeFiles/bevfusion.dir/src/main.cpp.o' failed
make[2]: *** [CMakeFiles/bevfusion.dir/src/main.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/bevfusion.dir/all' failed
make[1]: *** [CMakeFiles/bevfusion.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

hopef, 2023-05-26:
You must follow the steps described in the readme.
1. sudo apt-get install git-lfs
2. git clone --recursive https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution

This repo has a lot of dependencies that will be handled by the --recursive and git-lfs.

HZQNPU, 2023-05-26:
好的 非常感谢

closed 2023-05-26 by hopef

leijr, 2023-11-26:
为什么我按照这个操作进行完：
sudo apt-get install git-lfs
git clone --recursive https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution
还是一样的出现这个错误？

leijr, 2023-11-26:
大家出现这个问题的，记得多试几次。。。我试了两次好了