NVIDIA-AI-IOT/Lidar_AI_Solution #43: support of custom dataset

shuyuan-wang, 2023-05-26:
Hi:
Does your sparse convolution support inference with input shape N*4? When I'm trying to inference, it says 

[image]

hopef, 2023-05-26:
Sorry, the libspconv currently only supports nx5 input. We will add further support as needed and release it in the future.

shuyuan-wang, 2023-05-26:
Does spatial shape  have any limit?

hopef, 2023-05-26:
There are no limitations. It just requires more effort.

closed 2023-05-26 by shuyuan-wang

wangxu5, 2023-06-26:
Can you please update libsponv.so to be able to support the N*3 input？Thanks~ @hopef

hopef, 2023-07-07:
Ok, I will add it

qiaxigua, 2024-12-23:
Can you please update libsponv.so to be able to support the N*6 input？Thanks~

qiaxigua, 2024-12-23:
Can you please update libspconv.so to be able to support the N*3 input？Thanks~ @hopef

qiaxigua, 2024-12-23:
Can you please update libspconv.so to be able to support the N*6 input？Thanks~ @hopef