NVIDIA-AI-IOT/Lidar_AI_Solution #44: CUDA-BEVFusion build head.bbox model failed

wangxj2014, 2023-05-26:
Thanks for the great work. 

In CUDA-BEVFusion, when I run "bash tool/build_trt_engine.sh" to build engine file,  head.bbox model build failed, but have no error info.
Could you please help me check what the problem is?

hopef, 2023-05-26:
You can find the error messages generated during compilation in the build directory.

wangxj2014, 2023-05-26:
When i run " trtexec --onnx=model/resnet50int8/head.bbox.onnx --fp16 --inputIOFormats=fp16:chw, --outputIOFormats=fp16:chw,fp16:chw,fp16:chw,fp16:chw,fp16:chw,fp16:chw, --saveEngine=model/resnet50int8/build/head.bbox.plan --memPoolSize=workspace:2048 --verbose --dumpLayerInfo --dumpProfile --separateProfileRun --profilingVerbosity=detailed --exportLayerInfo=model/resnet50int8/build/head.bbox.json
" I got this:

[05/26/2023-17:27:02] [V] [TRT] Fastest Tactic: -3360065831133338131 Time: inf
[05/26/2023-17:27:02] [V] [TRT] Deleting timing cache: 712 entries, served 1305 hits since creation.
[05/26/2023-17:27:02] [E] Error[10]: [optimizer.cpp::computeCosts::2033] Error Code 10: Internal Error (Could not find any implementation for node {ForeignNode[(Unnamed Layer* 7) [Constant] + (Unnamed Layer* 8) [Shuffle]...Transpose_259 + (Unnamed Layer* 379) [Shuffle]]}.)
[05/26/2023-17:27:02] [E] Error[2]: [builder.cpp::buildSerializedNetwork::619] Error Code 2: Internal Error (Assertion engine != nullptr failed. )
[05/26/2023-17:27:02] [E] Engine could not be created from network
[05/26/2023-17:27:02] [E] Building engine failed
[05/26/2023-17:27:02] [E] Failed to create engine from model.
[05/26/2023-17:27:02] [E] Engine set up failed

hopef, 2023-05-29:
You may consider updating the TensorRT version to solve this issue.

wangxj2014, 2023-05-30:
I  successfully build engine using TensorRT = 8.5.0.

wangxj2014, 2023-05-30:
I successfully build engine using TensorRT = 8.5.0,  THANK YOU!

closed 2023-05-30 by hopef