NVIDIA-AI-IOT/Lidar_AI_Solution #48: how to train model/resnet50int8/bevfusion_ptq.pth ?

hzm8341, 2023-05-29:
I have train the code in bevfuion[@ db75150] ,using resnet50 config ,but when I export my pth to onnx, that is wrong.

hopef, 2023-05-29:
You can refer to the case we provided for ONNX export. If there is an error, please comment it here. Thanks

hzm8341, 2023-05-29:
you provided example pth file is goog to export onnx，but I want to use my trained pth file.
could you provid the train code or configs ?

hopef, 2023-05-30:
We did not change anything in the training code, and you can follow the steps in the readme to get your ONNX.

hzm8341, 2023-05-31:
OK,thanks,I will try again.

closed 2023-05-31 by hzm8341

OrangeJuice3211, 2024-06-20:
> OK,thanks,I will try again.

hello，did you solve this problem？ I want to use my trained pth file to export onnx，but there is no model in the pth file,there is state_dict only.

OrangeJuice3211, 2024-06-20:
> > OK,thanks,I will try again.
> 
> hello，did you solve this problem？ I want to use my trained pth file to export onnx，but there is no model in the pth file,there is state_dict only.

and the pth file is got from bevfusion train code.