NVIDIA-AI-IOT/Lidar_AI_Solution #49: How to set environment variables when I use pip install TensorRT?

BroPants, 2023-05-29:
Hi. I'm trying to use conda to run BEVFusion. I have installed tensorRT by pip and the function works. But I didn't find TensorRT_Lib and the other path under my conda env lib. Is there any way to solve this? THANK YOU.

[image]

hopef, 2023-05-30:
You can run the following script to find your TensorRT installation location:
$> `python -c "import tensorrt;import os;print(os.path.join(os.path.dirname(tensorrt.__file__), 'tensorrt.so'))" | xargs ldd | grep libnvinfer.so`

It will print the following message:
        libnvinfer.so.8 => **/path/to/TensorRT-8.5.3.1**/lib/libnvinfer.so.8 (0x00007f2aeb90a000)

**/path/to/TensorRT-8.5.3.1** is your TensorRT directory location.

BroPants, 2023-05-30:
Thanks hopef. I simply copy & paste the command you provided but it returns "No such file or directory" error.

However, I do find the <tensorrt.so> file in another directory named "tenssort_bindings". Is this the same file as you mentioned or I should install TensorRT C++ library additionally?