NVIDIA-AI-IOT/Lidar_AI_Solution #53: CenterPoint different inference result compared with Readme.md

bigsheep2018, 2023-05-31:
Hello Lidar_AI_Solution Team @hopef ,

Following the instructions mentioned in [Readme.md](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-CenterPoint/README.md), I can get the results as follows

[image]

**we can see the Detection NUM is different from the official result mentioned in Readme.md as follows**

[image]

**For further checking the difference, I visualize the results in rviz and find more false positive detection from my results**
**Official performance in Readme.md:**
[image]

**My visualization (ignore construction_vehicle, barriers, trailer, traffic_cone):**
[image]

**May I know why this situation happen? is it normal (maybe caused by hardware difference or different post-processing code)? and is there any solution to solve it?**

hopef, 2023-05-31:
Sorry, the data in the readme is legacy. 
We fixed bugs in the code, but we didn't update the readme in time.

bigsheep2018, 2023-05-31:
thanks @hopef . closing the issue.

closed 2023-05-31 by bigsheep2018