NVIDIA-AI-IOT/Lidar_AI_Solution #56: In Orin , come across library incompatibilities issue

JunLFang, 2023-05-31:
When I try to run command: 
bash tool/run.sh

I flashed Jpack5 into my Orin

find below error: Configuration done!
CMake Warning at /usr/share/cmake-3.16/Modules/FindProtobuf.cmake:499 (message):
  Protobuf compiler version doesn't match library version 3.6.1
Call Stack (most recent call first):
  CMakeLists.txt:31 (find_package)

-- Configuring done
-- Generating done
-- Build files have been written to: /home/hud/ssd/lidar_detection/Lidar_AI_Solution/CUDA-BEVFusion/build
[  4%] Linking CXX shared library libbevfusion_core.so
/usr/bin/ld:/home/hud/ssd/lidar_detection/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/aarch64/libspconv.so: file format not recognized; treating as linker script

Have you ever find this issue before?

hopef, 2023-05-31:
This may be caused by you not installing Git-lfs before git clone. libspconv.so is stored through lfs.
Consider manually downloading libspconv.so from GitHub instead.

JunLFang, 2023-06-01:
Thank you very much, I configured git-lfs , then cloned again , I found that this time , the size of libspconv.so is correct, and can pass the build

closed 2023-06-01 by JunLFang