NVIDIA-AI-IOT/Lidar_AI_Solution #58: pretrained model bevfusion-det.pth and onnx files

dpan817, 2023-06-01:
For the pretrained model bevfusion-det.pth, did you train it by youself, or download from other repository? How did you generate the onnx files in the same directory? Can I just run the scripts under /Lidar_AI_Solution/CUDA-BEVFusion/qat with bevfusion-det.pth to generate these onnx files?

```
model
├── resnet50
│   ├── **bevfusion-det.pth**
│   ├── camera.backbone.onnx
│   ├── camera.vtransform.onnx
│   ├── default.yaml
│   ├── fuser.onnx
│   ├── head.bbox.onnx
│   └── lidar.backbone.xyz.onnx
```

dpan817, 2023-06-01:
The pretrained model from MIT https://bevfusion.mit.edu/files/pretrained_updated/bevfusion-det.pth is different from the one in the model.zip

hopef, 2023-06-01:
For the pretrained model bevfusion-det.pth, did you train it by youself, or download from other repository?
-> `bevfusion-det.pth` is obtained by training with `resnet50/default.yaml`.

Can I just run the scripts under /Lidar_AI_Solution/CUDA-BEVFusion/qat with bevfusion-det.pth to generate these onnx files?
-> Yes, you can.

closed 2023-06-01 by hopef