NVIDIA-AI-IOT/Lidar_AI_Solution #66: libspconv.so: file format not recognized; treating as linker script

lishanxiuxiu, 2023-06-02:
### Prerequisites
To build the centerpoint inference, **CUDA**, **TensorRT** and **[libspconv](/libraries/3DSparseConvolution)** libraries are needed.
Note that libspconv only supports sm_80 & sm_86 on Tesla Platform and sm_87 on Embedded Platform.

As stated by reademe.md, Can't I use another graphics card to recognize libspconv.so?
How to solve the problem？

hopef, 2023-06-05:
You need to pull the code after installing `git-lfs`.

hopef, 2023-06-06:
I'm very sorry, I've removed the lfs due to too many problems caused by gitlfs. so you will not encounter this problem if you re-download the repo.

lishanxiuxiu, 2023-06-06:
The problem has been resolved, thank you！

closed 2023-06-06 by hopef

gauthamnarayan, 2023-09-06:
@hopef thanks for your help with all these issues. Do you think we can use libspconv for sm_80 or sm_86 on platforms other than Tesla? Basically we need to run this on 3090's and 4090's, is there anything we can do for this?

Also is there any possibility to release code for libspconv so that implement more custom tensorrt kernels?

hopef, 2023-09-06:
I think you can run libspconv well on 3090's and 4090's well.

krishnatoshniwal, 2023-09-06:
I get this error when trying to build on a 4090.

```
[09/05/2023-16:36:40] [E] Error[1]: [caskUtils.cpp::trtSmToCask::147] Error Code 1: Internal Error (Unsupported SM: 0x809)
[09/05/2023-16:36:40] [E] Error[2]: [builder.cpp::buildSerializedNetwork::609] Error Code 2: Internal Error (Assertion enginePtr != nullptr failed. )
[09/05/2023-16:36:40] [E] Engine could not be created from network
```

GPU has cuda devices: 2
----device id: 0 info----
  GPU : NVIDIA GeForce RTX 4090 
  Capbility: 8.9
  Global memory: 24217MB
  Const memory: 64KB
  SM in a block: 48KB
  warp size: 32
  threads in a block: 1024
  block dim: (1024,1024,64)
  grid dim: (2147483647,65535,65535)
----device id: 1 info----
  GPU : NVIDIA GeForce RTX 4090 
  Capbility: 8.9
  Global memory: 24217MB
  Const memory: 64KB
  SM in a block: 48KB
  warp size: 32
  threads in a block: 1024
  block dim: (1024,1024,64)
  grid dim: (2147483647,65535,65535)

krishnatoshniwal, 2023-09-06:
Updating the TensorRT version to 8.5 resolved the issue.