NVIDIA-AI-IOT/Lidar_AI_Solution #68: Result

chenzb0712, 2023-06-04:
Hello, I ran the code on orin according to the operation on github, but the running time is not as fast as you said, is there something wrong with my settings?
[image]

hopef, 2023-06-05:
There are large fluctuations in elapsed time, as indicated by your print message. You need to confirm that the device is working at a stable maximum performance mode.

closed 2023-06-08 by hopef

dav695, 2023-09-05:
Hello, 
In my case i have the same problem. My Total inference time is around 190 ms. I am using the device Jetson Orin developer Kit. I do not understand what i have to do to test this part:  "You need to confirm that the device is working at a stable maximum performance mode".
Thanks you so much!