NVIDIA-AI-IOT/Lidar_AI_Solution #69: CUDA-BEVFusion ./tool/run.sh problem

hunagjingwei, 2023-06-05:
bash ./tool/run.sh 
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: OFF
||
||  TensorRT: /usr/lib/aarch64-linux-gnu
||  CUDA: /usr/local/cuda-11.4
||  CUDNN: /usr/local/cuda-11.4/lib64
==========================================================
Try to get the current device SM
Current CUDA SM: 87
Configuration done!
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jtrobot/projects/Lidar_AI_Solution/CUDA-BEVFusion/build
[  4%] Linking CXX shared library libbevfusion_core.so
[ 63%] Built target bevfusion_core
[ 68%] Linking CXX executable bevfusion
[100%] Built target bevfusion
Create by resnet50int8, int8
------------------------------------------------------
Camerea Backbone 🌱 is Static Shape model
Inputs: 2
	0.img : {1 x 6 x 3 x 256 x 704} [Float16]
	1.depth : {1 x 6 x 1 x 256 x 704} [Float16]
Outputs: 2
	0.camera_depth_weights : {6 x 118 x 32 x 88} [Float16]
	1.camera_feature : {6 x 32 x 88 x 80} [Float16]
------------------------------------------------------
------------------------------------------------------
Camerea VTransform 🌱 is Static Shape model
Inputs: 1
	0.feat_in : {1 x 80 x 360 x 360} [Float16]
Outputs: 1
	0.feat_out : {1 x 80 x 180 x 180} [Float16]
------------------------------------------------------
------------------------------------------------------
Transfusion 🌱 is Static Shape model
Inputs: 2
	0.camera : {1 x 80 x 180 x 180} [Float16]
	1.lidar : {1 x 256 x 180 x 180} [Float16]
Outputs: 1
	0.middle : {1 x 512 x 180 x 180} [Float16]
------------------------------------------------------
------------------------------------------------------
BBox 🌱 is Static Shape model
Inputs: 1
	0.middle : {1 x 512 x 180 x 180} [Float16]
Outputs: 6
	0.reg : {1 x 2 x 200} [Float16]
	1.height : {1 x 1 x 200} [Float16]
	2.dim : {1 x 3 x 200} [Float16]
	3.rot : {1 x 2 x 200} [Float16]
	4.vel : {1 x 2 x 200} [Float16]
	5.score : {1 x 10 x 200} [Float16]
------------------------------------------------------
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 	0.93482 ms
[⏰ [NoSt] ImageNrom]: 	5.43446 ms
[⏰ Lidar Backbone]: 	33.61847 ms
[⏰ Camera Depth]: 	0.19194 ms
[⏰ Camera Backbone]: 	18.17446 ms
[⏰ Camera Bevpool]: 	4.22256 ms
[⏰ VTransform]: 	5.31846 ms
[⏰ Transfusion]: 	12.05306 ms
[⏰ Head BoundingBox]: 	21.66070 ms
Total: 95.240 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 	0.48627 ms
[⏰ [NoSt] ImageNrom]: 	6.10666 ms
[⏰ Lidar Backbone]: 	25.51798 ms
[⏰ Camera Depth]: 	0.17414 ms
[⏰ Camera Backbone]: 	16.71376 ms
[⏰ Camera Bevpool]: 	5.41965 ms
[⏰ VTransform]: 	2.91725 ms
[⏰ Transfusion]: 	9.39763 ms
[⏰ Head BoundingBox]: 	21.62912 ms
Total: 81.770 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 	0.48358 ms
[⏰ [NoSt] ImageNrom]: 	4.35056 ms
[⏰ Lidar Backbone]: 	23.81494 ms
[⏰ Camera Depth]: 	0.15594 ms
[⏰ Camera Backbone]: 	14.08912 ms
[⏰ Camera Bevpool]: 	1.82829 ms
[⏰ VTransform]: 	1.88509 ms
[⏰ Transfusion]: 	7.75245 ms
[⏰ Head BoundingBox]: 	12.79530 ms
Total: 62.321 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 	0.66467 ms
[⏰ [NoSt] ImageNrom]: 	5.13062 ms
[⏰ Lidar Backbone]: 	13.39149 ms
[⏰ Camera Depth]: 	0.10595 ms
[⏰ Camera Backbone]: 	10.58029 ms
[⏰ Camera Bevpool]: 	1.49952 ms
[⏰ VTransform]: 	1.69706 ms
[⏰ Transfusion]: 	5.55802 ms
[⏰ Head BoundingBox]: 	10.84301 ms
Total: 43.675 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 	0.66109 ms
[⏰ [NoSt] ImageNrom]: 	5.06966 ms
[⏰ Lidar Backbone]: 	13.44960 ms
[⏰ Camera Depth]: 	0.10109 ms
[⏰ Camera Backbone]: 	8.92582 ms
[⏰ Camera Bevpool]: 	1.47123 ms
[⏰ VTransform]: 	1.63341 ms
[⏰ Transfusion]: 	5.47510 ms
[⏰ Head BoundingBox]: 	11.14448 ms
Total: 42.201 ms
=============================================
==================BEVFusion===================
[⏰ [NoSt] CopyLidar]: 	0.66656 ms
[⏰ [NoSt] ImageNrom]: 	5.10880 ms
[⏰ Lidar Backbone]: 	13.24838 ms
[⏰ Camera Depth]: 	0.10240 ms
[⏰ Camera Backbone]: 	11.56803 ms
[⏰ Camera Bevpool]: 	1.49040 ms
[⏰ VTransform]: 	1.65869 ms
[⏰ Transfusion]: 	5.48061 ms
[⏰ Head BoundingBox]: 	13.41373 ms
Total: 46.962 ms
=============================================
./tool/run.sh: line 41:  7985 Segmentation fault      (core dumped) ./build/bevfusion $DEBUG_DATA $DEBUG_MODEL $DEBUG_PRECISION

hunagjingwei, 2023-06-05:
> This is caused by the font file exception in the visualization function. You need to pull the code after installing `git-lfs`. Or you need to make sure the `simhei.ttf` file is correct.

I already install 'git-lfs' before pull the code ;and how to determine `simhei.ttf` file is correct

hopef, 2023-06-05:
Could you run `ls -al tool/simhei.ttf` and send me the output?

hunagjingwei, 2023-06-06:
> ls -al tool/simhei.ttf

-rw-rw-r-- 1 jtrobot jtrobot 132 6月   2 15:15 tool/simhei.ttf

hopef, 2023-06-06:
This `simhei.ttf` file is not downloaded in full, you may consider git pull to update it again.

I'm very sorry, I've removed the `git-lfs` due to too many problems caused by that. So you will avoid this problem if you re-download this repo.

hunagjingwei, 2023-06-06:
> This `simhei.ttf` file is not downloaded in full, you may consider git pull to update it again.
> 
> I'm very sorry, I've removed the `git-lfs` due to too many problems caused by that. So you will avoid this problem if you re-download this repo.

thanks for your help,I redownload this repo and this problem is solved .

closed 2023-06-06 by hopef