NVIDIA-AI-IOT/Lidar_AI_Solution #70: [CUDA-Centerpoint]  convert .pth to 。.onnx

chenxyyy, 2023-06-05:


chenxyyy, 2023-06-05:
This is my understanding of the process of using `CUDA-Centerpoint`
1. train dataset，get the weight.
2. `python export-scn.py` get `centerpoint.scn.onnx`; `python export_neck_head.py` get `rpn_centerhead_sim.onnx`
3. build trt model
4. build the `CUDA-Centerpoint`

hopef, 2023-06-06:
When constructing onnx node, the attribute value of None is not allowed.
Your process is correct. But your pth file may be different from the one we generated.

chenxyyy, 2023-06-06:
This `pth` file was trained on Nuscense using the [centerpoint project](https://github.com/tianweiy/CenterPoint)

The commond shows below. 

```
python tools/create_data.py nuscenes_data_prep --root_path="/path/to/nuScenes-mini/" --version="v1.0-mini" --nsweeps=10

python ./tools/train.py configs/nusc/voxelnet/nusc_centerpoint_voxelnet_0075voxel_fix_bn_z.py
```
Is there any problem with this operation?

hopef, 2023-06-06:
Please help me to check which of the `xxx` attributes mentioned in the following code `self.xxx` in your model is `None`?

```python  
nodes.append(
      helper.make_node(
          "SparseConvolution", inputs, [get_tensor_id(y)], f"conv{ilayer}", 
          ndim = self.ndim,
          input_spatial_shape = x.spatial_shape,
          output_spatial_shape = y.spatial_shape,
          in_channels = self.in_channels,
          out_channels = self.out_channels,
          kernel_size = self.kernel_size,
          output_bound = output_bound,
          stride = self.stride,
          dilation = self.dilation,
          padding = self.padding,
          transposed = self.transposed,
          inverse = self.inverse,
          output_padding = self.output_padding,
          groups = self.groups,
          subm = self.subm,
          rulebook = self.indice_key,
          activation = act_type_name[self.act_type],
          input_shape  = x.features.shape,
          output_shape = y.features.shape
      )
  )
```

chenxyyy, 2023-06-06:
Yes. `self.indice_key` is `None`.

I have tried the `pth` file  [epoch_20.pth](https://mitprod-my.sharepoint.com/:f:/g/personal/tianweiy_mit_edu/EhgzjwV2EghOnHFKyRgSadoBr2kUo7yPu52N-I3dG3c5dA?e=a9MdhX) offered in `3DSparseConvolution` module. It didn't work either.

hopef, 2023-06-06:
Maybe you can modify the following code to solve this problem.

```python
nodes.append(
      helper.make_node(
          "SparseConvolution", inputs, [get_tensor_id(y)], f"conv{ilayer}", 
          ndim = self.ndim,
          input_spatial_shape = x.spatial_shape,
          output_spatial_shape = y.spatial_shape,
          in_channels = self.in_channels,
          out_channels = self.out_channels,
          kernel_size = self.kernel_size,
          output_bound = output_bound,
          stride = self.stride,
          dilation = self.dilation,
          padding = self.padding,
          transposed = self.transposed,
          inverse = self.inverse,
          output_padding = self.output_padding,
          groups = self.groups,
          subm = self.subm,
          rulebook = self.indice_key if self.indice_key else "",
          activation = act_type_name[self.act_type],
          input_shape  = x.features.shape,
          output_shape = y.features.shape
      )
  )
```

chenxyyy, 2023-06-06:
Very good, the `CUDA-Centerpoint` process has been fully completed. Next, I will try to test the accuracy on Nuscense-dataset.

Thank you.

closed 2023-06-06 by chenxyyy

hopef, 2023-06-06:
It is worth noting that the `submanifold convolution` must care about the `indice_key` setting. 
Because it can lead to a `9 ms` performance improvement on ORIN.

kellenf, 2023-10-09:
Can you show your onnx graph? thanks!

mx2013713828, 2024-07-01:
hello,can you share some advices about export .pth to .onnx? 
Do I have to use the parameter "nusc_centpoint_voxelnet_0075voxel_fix_bn_z" for training? 
I want to train a custom model with custom dataset, can you give me some advices about the pipeline?
thank you very much!