NVIDIA-AI-IOT/Lidar_AI_Solution #71: Compute Capability can not be found correctly in JetPack5.1.1 container

hren20, 2023-06-06:
I am trying to deploy BEVFusion on Orin. I can run the code correctly in bare metal instead of container. 
My bare metal installs JetPack5.0.2, Its effect is relatively poor. So I want run the code in JetPack5.1.1. I use NVIDIA offical JetPack5.1.1 image. When I run environment.sh, this error occurs. I have tried to copy libcuda.so to /usr/lib/aarch64-linux-gnu, but Compute Capability still can not find correctly.
I located the problem to tool/cudasm.sh, and the return value of ’ret’ is 34, which should be 0 under normal circumstances. How can I solve this problem? Many thanks.
```
. tool/environment.sh 
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: OFF
||
||  TensorRT: /lib/aarch64-linux-gnu
||  CUDA: /usr/local/cuda
||  CUDNN: /usr/lib/aarch64-linux-gnu/
==========================================================
Try to get the current device SM
Traceback (most recent call last):
  File "<string>", line 4, in <module>
  File "/usr/lib/python3.8/ctypes/__init__.py", line 451, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib/python3.8/ctypes/__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [libcuda.so](http://libcuda.so/): cannot open shared object file: No such file or directory
Current CUDA SM: 
Configuration done!
```

hopef, 2023-06-06:
If libcuda.so is not found, you will encounter other following errors.
You can avoid this error by removing the automatically find `sm` code.

hren20, 2023-06-06:
> If libcuda.so is not found, you will encounter other errors that follow. You can avoid this error by removing the automatically find `sm` code.

I have tried to specify cudasm instead of getting it automatically. When I run `bash tool/build_trt_engine.sh`, the program runs too fast. When I look at the` /model/resnet50int8/build` directory, I find that there is no engine generated. The image I used is `nvcr.io/nvidia/l4t-jetpack:r35.3.1`. Is there anything that needs extra attention?
```
bash tool/build_trt_engine.sh 
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: OFF
||
||  TensorRT: /lib/aarch64-linux-gnu
||  CUDA: /usr/local/cuda
||  CUDNN: /usr/lib/aarch64-linux-gnu/
==========================================================
Try to get the current device SM
Current CUDA SM: 87
Configuration done!
Building the model: model/resnet50int8/build/camera.backbone.plan, this will take several minutes. Wait a moment 🤗🤗🤗~.
Building the model: model/resnet50int8/build/fuser.plan, this will take several minutes. Wait a moment 🤗🤗🤗~.
Building the model: model/resnet50int8/build/camera.vtransform.plan, this will take several minutes. Wait a moment 🤗🤗🤗~.
Building the model: model/resnet50int8/build/head.bbox.plan, this will take several minutes. Wait a moment 🤗🤗🤗~.
```

hopef, 2023-06-06:
You can check the log in the `model/resnet50int8/build/` directory. it should be encountering other errors.
For the latest commits, I've added error checking, which will indicate to you that an error has occurred.

hren20, 2023-06-06:
> You can check the log in the `model/resnet50int8/build/` directory. it should be encountering other errors. For the latest commits, I've added error checking, which will indicate to you that an error has occurred.

Thanks a lot. The log file shows that there is no `libnvdla_compiler.so` file. Then I found that there is no such folder of `/usr/lib/aarch64-linux-gnu/tegra`. The image I used is `nvcr.io/nvidia/l4t-jetpack:r35.3.1`. This fold can be found in my bare metal that installed JetPack5.0.2. Would you please give me some advice about this?
```
# here is the content of log file in model/resnet50int8/build/
trtexec: error while loading shared libraries: [libnvdla_compiler.so](http://libnvdla_compiler.so/): cannot open shared object file: No such file or directory
```

hren20, 2023-06-08:
I have solved this problem, it is a rookie mistake. The folder`/usr/lib/aarch64-linux-gnu/tegra` is missing because the relevant settings of nvidia-container-runtime are not added in docker's configuration file. Refer to [nvidia offical document](https://github.com/nvidia/nvidia-container-runtime) to modify `ExecStart =/usr/bin/dockerd-add-runtime=nvidia=/usr/bin/nvidia-container-runtime` or use the command `sudo dockerd-add-runtime=nvidia=/usr/bin/nvidia-container-runtime [...]` to add nvidia-container-runtime to the docker.

closed 2023-06-08 by hren20