NVIDIA-AI-IOT/Lidar_AI_Solution #78: CUDA_BEVFUSION :convert transhead.onnx  to engine in orin meet ScatterND_49 error

wutheringcoo, 2023-06-07:
@hopef 
when i convert transhead.onnx  to engine in orin, i meet an error: Internal error(Assertion !layerImpls.empty()) failed. ScatterND_49 has no runners that implement it. how can i fix this error?
env tensorrt : 8.4.10

hopef, 2023-06-07:
Consider upgrading the TensorRT version to 8.5 or higher. Or use a plugin to support it.

wutheringcoo, 2023-06-07:
Hello, The attachment is log.
[head.bbox.log](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/files/11676563/head.bbox.log)

hren20, 2023-06-08:
> When I convert the same onnx to engine in x86[TensorRT version is 8.4.0], it success! And i see the log both in orin and x86: "Registered plugin creator - :: ScatterND version 1", which means supporting ScatterND? so i guess the reason is not miss ScatterND plugin?

I have tried bevfusin in orin by using jetpack5.0.2 whcih has installed tensorrt8.4.1. The convertion can be run, but the inference result is bad. The result is great by using tensorrt8.5.2 with jetpack5.1.1. So I sincerely recommend you to update your TensorRT version.

hopef, 2023-06-08:
It looks like a solution can only be obtained by updating TensorRT. 
We also encountered the `transformer` inference value error in lower TensorRT.

closed 2023-06-09 by hopef