NVIDIA-AI-IOT/Lidar_AI_Solution #80: How to transform local data to BEVFusion example-data

ryontang, 2023-06-09:
Hi, I wanna use the local data to do infer, but I am wondering how to transform local date to tensor file.
From the [#74](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/issues/74), it has mentioned [tool/dump-data.py](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/tool/dump-data.py), but how to get the dimention of tensor file so that we could do the next transformation, could someone give an example?

hopef, 2023-06-10:
As you know, the tensor file is a very simple format. It just stores the `magic number/dtype/shape/data` information. You can see the C++ read code [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/src/common/tensor.cu#L506). See the python read code [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/lean/tensor.py#L49). My recommendation is that you should read this part of the code.

ryontang, 2023-06-12:
> As you know, the tensor file is a very simple format. It just stores `magic number/dtype/shape/data` only. You can see the C++ read code [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/src/common/tensor.cu#L506). See the python read code [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/qat/lean/tensor.py#L49). My recommendation is that you should read this part of the code.

okay, thanks a lot!

closed 2023-06-12 by ryontang