NVIDIA-AI-IOT/Lidar_AI_Solution #82: The indice from scn engine is incorrect

Junx0924, 2023-06-09:
Hi,
Thanks for this fantastic work.

I find the indices [(pyscn.cpp)](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/libraries/3DSparseConvolution/src/pyscn.cpp) are not correctly extracted from scn engine.

numel is the size of the out_indice, however it is 1.
ndim is the dimension of the out_indice, it is 0.
Can you help to solve this issue?

[image]

hopef, 2023-06-10:
For the SCN model, the last layer converts the sparse tensor to dense. 
therefore it will only have feature(out_feature), but no indices(out_indices). So the result I observed here is correct.

Junx0924, 2023-06-11:
So this is the result of the current scn onnx model.

Can you show a way to add indices as an output to the onnx model?

hopef, 2023-06-12:
You will get indices if you remove the `ScatterToDense` node in the scn model.

Junx0924, 2023-06-12:
Thanks, after modifying the scn onnx model, now it looks correct!

[image]
[image]

closed 2023-06-12 by Junx0924