NVIDIA-AI-IOT/Lidar_AI_Solution #83: Hello, may I ask if there are any reference materials available for the sparse convolution using ONNX for inference in bevfusion?

windkiss5, 2023-06-11:
您好，请问bevfusion中稀疏卷积使用onnx去推理，这部分有什么参考资料嘛？

十分感谢你们的工作！我是一名tensorrt的初学者，对于spconv这部分的推理，我查看其他issue发现，是“It is not a plugin. Because the data-depend characteristic can not be supported in the current version of TensorRT. So we implemented a tiny engine in a standalone way.”，但是我不太明白具体实现的原理，请问方便提供一点学习的方向嘛？

Hello, thank you very much for your kind works. As for the inference part of spconv, I have found in other issues that "It is not a plugin. Because the data-depend characteristic cannot be supported in the current version of TensorRT. So we implemented a tiny engine in a standalone way." However, I do not completely understand the specific implementation principle. Could you please provide some guidance for me to learn more about it?

Junx0924, 2023-06-11:
“It is not a plugin. Because the data-depend characteristic can not be supported in the current version of TensorRT. So we implemented a tiny engine in a standalone way.”

Because the current version of TensorRT only supports inputs and outputs with fixed dimensions.
In the sparse Conv, the output can't be known in advance. Only after the calculation, you will know the dimensions of the output.
That's why it can't be added as a plugin in the current TensorRT.

windkiss5, 2023-06-12:
Thank you for your response. As a beginner, I'm still not quite clear. So in the unopened code for the lidar branch, is the "built small model" mainly inferred using ONNX Runtime? Or is there another way using TensorRT?

hopef, 2023-06-13:
This core is based on some custom CUDA kernel.
ONNX is only used as a weight and structure store.

closed 2023-06-15 by hopef