NVIDIA-AI-IOT/Lidar_AI_Solution #85: May I use my libspconv.so

EmmanuelAntonioC, 2023-06-12:
Hi! Thanks for this amazing repo. 
I'm trying to do some test with the CenterPoint model and I wanted to know if I can use a different `libspconv.so`. The `libspconv.so` that I'm trying to use is the one obtained with in the repo https://github.com/traveller59/spconv. I tried to change it but I'm having this issue:

```
/usr/bin/ld: CMakeFiles/centerpoint.dir/src/centerpoint.cpp.o: in function `CenterPoint::CenterPoint(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool)':
centerpoint.cpp:(.text+0x2306): undefined reference to `spconv::load_engine_from_onnx(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, spconv::Precision)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/centerpoint.dir/build.make:763: centerpoint] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/centerpoint.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

I tried adding the `3rdparty/libspconv/include/spconv` `.hpp` files in the `MakeLists.txt` but still doesn't work. Do you know if it is possible to use a different `libspconv.so` or if I missing something?
 thanks in advance :)

hopef, 2023-06-13:
This error is caused by a missing link to `libspconv.so`.
You need to add `libspconv.so` to link flags to fix this issue.

EmmanuelAntonioC, 2023-06-13:
Thanks for your reply. I am trying to use the `libspconv.so` obtained with the https://github.com/traveller59/spconv. 
Do you know if I need to do something to make it work?

hopef, 2023-06-14:
We provided the CenterPoint example. It includes how to use the model generated by traveller59/spconv. You can refer to it.

closed 2023-06-15 by hopef

skprot, 2023-08-06:
> Thanks for your reply. I am trying to use the `libspconv.so` obtained with the https://github.com/traveller59/spconv. Do you know if I need to do something to make it work? @EmmanuelAntonioC 

Did you solved this issue?

EmmanuelAntonioC, 2023-08-07:
@skprot No, I couldn't solve this, seems that the `libspconv.so` provided in this repo has some differences with the one compiled from https://github.com/traveller59/spconv.

vardeep-motor-ai, 2024-01-11:
Did you manage to solve this issue?

riteshkhrn, 2024-09-25:
Try this implementation it is a bridge between libspconv.so and Lidar_AI_Solution. 

https://github.com/riteshkhrn/Lidar_AI_Solution/tree/main/libraries/New3DSparseConvolution