NVIDIA-AI-IOT/Lidar_AI_Solution #86: Cmake Error

alre5639, 2023-06-12:
I get the following error when running `bash tool/run.sh`

```
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: OFF
||
||  TensorRT: /usr/lib/x86_64-linux-gnu
||  CUDA: /usr/local/cuda-12.1/
||  CUDNN: /usr/lib/x86_64-linux-gnu
==========================================================
Try to get the current device SM
Current CUDA SM: 86
Configuration done!
-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /home/alec/Lidar_AI_Solution/CUDA-BEVFusion/build
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/common/bevfusion_core_generated_tensor.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-backbone.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-bevpool.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-depth.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-geometry.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-normalization.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-vtransform.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_head-transbbox.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_lidar-voxelization.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_transfusion.cu.o
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
CMake Error at bevfusion_core_generated_camera-depth.cu.o.Release.cmake:280 (message):
  Error generating file
  /home/alec/Lidar_AI_Solution/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/bevfusion/./bevfusion_core_generated_camera-depth.cu.o

make[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:909: CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-depth.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
/usr/include/c++/11/type_traits:79:52: error: redefinition of ‘constexpr const _Tp std::integral_constant<_Tp, __v>::value’
   79 |   template<typename _Tp, _Tp __v>
      |                                                    ^                           
/usr/include/c++/11/type_traits:67:29: note: ‘constexpr const _Tp value’ previously declared here
   67 |       static constexpr _Tp                  value = __v;
      |                             ^~~~~
CMake Error at bevfusion_core_generated_camera-backbone.cu.o.Release.cmake:280 (message):
  Error generating file
  /home/alec/Lidar_AI_Solution/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/bevfusion/./bevfusion_core_generated_camera-backbone.cu.o

make[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:352: CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-backbone.cu.o] Error 1
[... 351 lines cut ...]
  420 |   static constexpr int ITEMS_PER_THREAD = Policy::ITEMS_PER_THREAD;
      |                  ^~~~~~
/usr/local/cuda-12.1/include/cub/agent/agent_merge_sort.cuh:421:18: error: inline variables are only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Werror]
  421 |   static constexpr int ITEMS_PER_TILE = Policy::ITEMS_PER_TILE;
      |                  ^~~~~~
/usr/local/cuda-12.1/include/cub/agent/agent_merge_sort.cuh:422:18: error: inline variables are only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Werror]
  422 |   static constexpr int SHARED_MEMORY_SIZE =
      |                  ^~~~~~
/usr/local/cuda-12.1/include/cub/device/dispatch/dispatch_merge_sort.cuh:453:18: error: inline variables are only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Werror]
  453 |   static constexpr bool KEYS_ONLY = std::is_same<ValueT, NullType>::value;
      |                  ^~~~~~
cc1plus: all warnings being treated as errors
CMake Error at bevfusion_core_generated_camera-geometry.cu.o.Release.cmake:280 (message):
  Error generating file
  /home/alec/Lidar_AI_Solution/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/bevfusion/./bevfusion_core_generated_camera-geometry.cu.o

make[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:1726: CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-geometry.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/bevfusion_core.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```

I think this may be being caused by my enviornment setup, in the environment.sh file the locations of `TensorRT_Lib` and `TensorRT_Inc`. are proving diffuclt to determine. i have tried following the instuctions here: https://docs.nvidia.com/deeplearning/tensorrt/install-guide/index.html#installing-tar, trying both the debian installation and tar installation, but have gotten the same error everytime. with the tar installation lib and inc folder locations are quite obvious, the debian install I cant find where those would be, might be under a different name like lib-nvinfer* but not sure. 

I would appraciate any ideas on the error that I am getting. Thanks!

alre5639, 2023-06-12:
I think this is similar to the error where the wrong c++ language version was used similar to https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/issues/21 I can compile with g++ -std=c++17 for single files but dont know how to specify the c++ version for the `bash tool/run.sh` command

alre5639, 2023-06-12:
by changing the c++ compiler version from 14 to 17 in the cmakeLists.txt I am able to get the build to 60% now but get an error with the model definition:
```
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: OFF
||
||  TensorRT: /lib/x86_64-linux-gnu
||  CUDA: /usr/local/cuda-12.1
||  CUDNN: /usr/lib/x86_64-linux-gnu
==========================================================
Try to get the current device SM
Current CUDA SM: 86
Configuration done!
-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /home/alec/Lidar_AI_Solution/CUDA-BEVFusion/build
[  9%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-backbone.cu.o
[  9%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/common/bevfusion_core_generated_tensor.cu.o
[ 13%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-bevpool.cu.o
[ 18%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-depth.cu.o
[ 22%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-geometry.cu.o
[ 27%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-normalization.cu.o
[ 31%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-vtransform.cu.o
[ 36%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_lidar-voxelization.cu.o
[ 40%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_head-transbbox.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_transfusion.cu.o
[ 50%] Building CXX object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion.cpp.o
[ 54%] Building CXX object CMakeFiles/bevfusion_core.dir/src/bevfusion/lidar-scn.cpp.o
[ 59%] Building CXX object CMakeFiles/bevfusion_core.dir/src/common/tensorrt.cpp.o
In file included from /home/alec/Lidar_AI_Solution/CUDA-BEVFusion/src/bevfusion/lidar-scn.cpp:24:
/home/alec/Lidar_AI_Solution/CUDA-BEVFusion/src/bevfusion/lidar-scn.hpp:39:15: error: field ‘model’ has incomplete type ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’}
   39 |   std::string model;
      |               ^~~~~
In file included from /usr/include/c++/11/iosfwd:39,
                 from /usr/include/c++/11/bits/shared_ptr.h:52,
                 from /usr/include/c++/11/memory:77,
                 from /home/alec/Lidar_AI_Solution/CUDA-BEVFusion/src/bevfusion/lidar-voxelization.hpp:27,
                 from /home/alec/Lidar_AI_Solution/CUDA-BEVFusion/src/bevfusion/lidar-scn.hpp:29,
                 from /home/alec/Lidar_AI_Solution/CUDA-BEVFusion/src/bevfusion/lidar-scn.cpp:24:
/usr/include/c++/11/bits/stringfwd.h:74:11: note: declaration of ‘std::string’ {aka ‘class std::__cxx11::basic_string<char>’}
   74 |     class basic_string;
      |           ^~~~~~~~~~~~
make[2]: *** [CMakeFiles/bevfusion_core.dir/build.make:3678: CMakeFiles/bevfusion_core.dir/src/bevfusion/lidar-scn.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/bevfusion_core.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

hopef, 2023-06-13:
Sorry, I need to update it.

hopef, 2023-06-13:
Set the C++ syntax options, which can be found [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/95ff5dea9eb486ef969a972939a5ef69bbc94f88/CUDA-BEVFusion/CMakeLists.txt#LL33C36-L33C36).

Sorry, The current `libspconv.so` is compiled on `CUDA11`.

alre5639, 2023-06-13:
No problem, thanks for the confirmation and the repo, I am workking on getting CUDA 11.x running to run the framework.

YoushaaMurhij, 2023-06-23:
@hopef, Is it possible to run it on cuda 10.X?. I want to try cuda-centerpoint on Jetson Xavier.

Thanks!

Li-Whasaka, 2023-10-18:
I meet the same error,have you solved it?

ryanyej, 2023-12-22:
same error . change c++14 to c++17

JYS997760473, 2024-02-05:
> Set the C++ syntax options, which can be found [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/95ff5dea9eb486ef969a972939a5ef69bbc94f88/CUDA-BEVFusion/CMakeLists.txt#LL33).
> 
> Sorry, The current `libspconv.so` is compiled on `CUDA11`.

Hi, I am still stuck in this kind of problem. My CUDA is 11.8, after using your new cmake syntax provided, it still shows the similar error

JYS997760473, 2024-02-05:
> I switch gcc-11 to gcc-9, and this problem is resolved

Hi, Could you please provide some hints about how to switch gcc-11 to gcc-9? I tried to switch gcc to 9 but when I go back to rebuild the project, it shows these errors:
```
==========================================================
||  MODEL: resnet50int8
||  PRECISION: int8
||  DATA: example-data
||  USEPython: OFF
||
||  TensorRT: /usr/lib/x86_64-linux-gnu
||  CUDA: /usr/local/cuda-11.8
||  CUDNN: /usr/lib/x86_64-linux-gnu
==========================================================
Try to get the current device SM
Current CUDA SM: 89
Configuration done!
-- The C compiler identification is GNU 9.5.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/gcc-9 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-11.8 (found version "11.8") 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found version "3.12.4") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jiayansong/workspace/Lidar_AI_Solution/CUDA-BEVFusion/build
[  4%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/common/bevfusion_core_generated_tensor.cu.o
[  9%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-backbone.cu.o
[ 13%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-depth.cu.o
[ 18%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-geometry.cu.o
[ 22%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-bevpool.cu.o
[ 27%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-normalization.cu.o
[ 31%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_camera-vtransform.cu.o
[ 36%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_head-transbbox.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_transfusion.cu.o
[ 45%] Building NVCC (Device) object CMakeFiles/bevfusion_core.dir/src/bevfusion/bevfusion_core_generated_lidar-voxelization.cu.o
gcc-9: fatal error: cannot execute ‘cc1plus’: execvp: No such file or directory
compilation terminated.
gcc-9: fatal error: cannot execute ‘cc1plus’: execvp: No such file or directory
compilation terminated.
nvcc fatal   : Failed to preprocess host compiler properties.
nvcc fatal   : Failed to preprocess host compiler properties.
gcc-9: fatal error: cannot execute ‘cc1plus’: execvp: No such file or directory
compilation terminated.
CMake Error at bevfusion_core_generated_tensor.cu.o.Release.cmake:220 (message):
  Error generating
  /home/jiayansong/workspace/Lidar_AI_Solution/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/common/./bevfusion_core_generated_tensor.cu.o

CMake Error at bevfusion_core_generated_camera-backbone.cu.o.Release.cmake:220 (message):
  Error generating
  /home/jiayansong/workspace/Lidar_AI_Solution/CUDA-BEVFusion/build/CMakeFiles/bevfusion_core.dir/src/bevfusion/./bevfusion_core_generated_camera-backbone.cu.o

```