NVIDIA-AI-IOT/Lidar_AI_Solution #87: FPS does not reach the predicted value

pocky111, 2023-06-13:
Hello, I tried to run the bev fusion model on orin. My environment configuration is Jetpack5.0.2, CUDA11.4.2, cuDNN8.4.1.50 and Tensorrt8.4.1.5. However, I got the following results after running the demo with the Resnet50PTQ model, the FPS value basically remained around 12-15, and my physical computer was in the highest power consumption mode. Can you give me some advice on this problem? thank you

[image]

hopef, 2023-06-14:
Your device seems to be unstable. Consider running it for a long time to see if it stabilizes.

pocky111, 2023-06-14:
Okay thanks,i' ll try it and see how it works.

closed 2023-06-15 by hopef