NVIDIA-AI-IOT/Lidar_AI_Solution #91: dump-data example

alre5639, 2023-06-14:
It would be incredibly helpful to provide an example as to how generate your own data using dump-data.py I am having a lot of trouble understanding how to get data into function correctly. Perhaps a quick example from the nuscenes sample data (https://www.nuscenes.org/data/v1.0-mini.tgz) would be possible?

hopef, 2023-06-14:
step1, run `python bevfusion/tools/create_data.py nuscenes --root-path /data/nuscenes/mini --out-dir /data/nuscenes/mini --extra-tag nuscenes --version=v1.0-mini`

step2, change the dataset_root to `/data/nuscenes/mini` in `bevfusion/configs/nuscenes/default.yaml`

step3, run `python tool/dump-data.py`

alre5639, 2023-06-15:
perfect thank you! For anyone wanted to perform step one it relies on the MIT-BEVfusion dependencies so I would recommend running it in their docker container found [here](https://github.com/mit-han-lab/bevfusion/tree/db75150717a9462cb60241e36ba28d65f6908607). just a note I did run into a bug in their docker when running create_data.py where the numba and numpy versions were missmatched so i just ran a quick `pip install numba==0.57.0`

closed 2023-06-15 by alre5639

JQZhai, 2023-07-04:
> step1, run `python bevfusion/tools/create_data.py nuscenes --root-path /data/nuscenes/mini --out-dir /data/nuscenes/mini --extra-tag nuscenes --version=v1.0-mini`
>
> step2, change the dataset_root to [...]

I am facing a problem like this
`ImportError: cannot import name 'build_dataloader' from 'mmdet3d.datasets'`
how to solve it, thanks