NVIDIA-AI-IOT/Lidar_AI_Solution #96: problem about  cuda sm

keristen, 2023-06-16:
my run not export CUDA sm, but this is no error?
please why???
[image]

hopef, 2023-06-16:
You can remove the code to get the SM in the script. Replace it with the manually set SM.

keristen, 2023-06-16:
Thank you very much. Yes, that's what I did, but I'm curious why there's no output for this, it looks very uncomfortable haha
Is there any possible reason

hopef, 2023-07-06:
This script calls the API to get sm after [loading libcuda.so](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-BEVFusion/tool/cudasm.sh#L26) using python. It may not load due to missing a symbol link of libcuda.so.