NVIDIA-AI-IOT/Lidar_AI_Solution #97: CUDA-BEVFusion : Sample data detection results are inaccurate

insmod-he, 2023-06-16:
It seems that the detection results are relatively poor. You can see that there is missed pedestrian in the back-left image. And the 3d box for the car in the back-image is inaccurate.

Besides, running the program multiple times, the outputs are different each time.

Are these normal?

[image]
[image]

> $ bash ./tool/run.sh
==========================================================
||  MODEL: resnet50
||  PRECISION: fp16
||  DATA: example-data
||  USEPython: OFF
||
||  TensorRT: /lib/x86_64-linux-gnu
||  CUDA: /usr/local/cuda-11.3
||  CUDNN: /usr/lib/x86_64-linux-gnu
==========================================================
Try to get the current device SM 
Current CUDA SM: 86
Configuration done!
CUDA_lib:/usr/local/cuda-11.3/lib64
TensorRT_Lib:/lib/x86_64-linux-gnu
-- Configuring done
-- Generating done
-- Build files have been written to: Lidar_AI_Solution/CUDA-BEVFusion/build
[ 63%] Built target bevfusion_core
[100%] Built target bevfusion
example-data resnet50 fp16
Create by resnet50, fp16 

...

>==================BEVFusion===================  
[⏰ [NoSt] CopyLidar]:  0.24701 ms              
[⏰ [NoSt] ImageNrom]:  2.00045 ms              
[⏰ Lidar Backbone]:    5.04730 ms              
[⏰ Camera Depth]:      0.02534 ms              
[⏰ Camera Backbone]:   4.49024 ms              
[⏰ Camera Bevpool]:    0.33274 ms              
[⏰ VTransform]:        0.56422 ms              
[⏰ Transfusion]:       2.08067 ms              
[⏰ Head BoundingBox]:  2.59891 ms              
Total: 15.139 ms                                
=============================================

hopef, 2023-06-16:
This abnormally. Could you tell me what operation you did? And what is the environment you are running in?

insmod-he, 2023-06-19:
Sorry for the late reply.

Environment:

```bash
# ubuntu
lsb_release -a
    No LSB modules are available.
    Distributor ID: Ubuntu
    Description:    Ubuntu 20.04.6 LTS
    Release:        20.04
    Codename:       focal

# python
python3
    Python 3.8.10 (default, May 26 2023, 14:05:08) 
    [GCC 9.4.0] on linux
    Type "help", "copyright", "credits" or "license" for more information.

# cuda
nvcc --version
    nvcc: NVIDIA (R) Cuda compiler driver
    Copyright (c) 2005-2021 NVIDIA Corporation
    Built on Mon_May__3_19:15:13_PDT_2021
    Cuda compilation tools, release 11.3, V11.3.109
    Build cuda_11.3.r11.3/compiler.29920130_0

# cudnn
cat /usr/include/cudnn_version.h | grep CUDNN_MAJOR -A 2 
    #define CUDNN_MAJOR 8
    #define CUDNN_MINOR 9
    #define CUDNN_PATCHLEVEL 0
    --
    #define CUDNN_VERSION (CUDNN_MAJOR * 1000 + CUDNN_MINOR * 100 + CUDNN_PATCHLEVEL)

ll /usr/lib/x86_64-linux-gnu/libcudnn.so.8
    lrwxrwxrwx 1 root root 17 Apr  5 16:06 /usr/lib/x86_64-linux-gnu/libcudnn.so.8 -> libcudnn.so.8.9.0

# TensorRT
/usr/src/tensorrt/bin/trtexec                                                                                                                                                                             
    &&&& RUNNING TensorRT.trtexec [TensorRT v8601] # /usr/src/tensorrt/bin/trtexec
    ...

# environment.sh
cat ./tool/environment.sh
    export TensorRT_Lib=/lib/x86_64-linux-gnu
    export TensorRT_Inc=/usr/include/x86_64-linux-gnu
    export TensorRT_Bin=/usr/src/tensorrt/bin

    export CUDA_Lib=/usr/local/cuda-11.3/lib64
    export CUDA_Inc=/usr/local/cuda-11.3/include
    export CUDA_Bin=/usr/local/cuda-11.3/bin
    export CUDA_HOME=/usr/local/cuda-11.3

    #export CUDNN_Lib=/path/to/cudnn/lib
    CUDNN_Lib=/usr/lib/x86_64-linux-gnu

    # resnet50/resnet50int8/swint
    export DEBUG_MODEL=resnet50

    # fp16/int8
    export DEBUG_PRECISION=fp16
    export DEBUG_DATA=example-data
    export USE_Python=OFF
    ...

# so
$ ldd ./build/libbevfusion_core.so 
    linux-vdso.so.1 (0x00007ffd7a328000)
    libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007efe9c9fb000)
    libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007efe9c9f5000)
    librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007efe9c9eb000)
    libnvinfer.so.8 => /lib/x86_64-linux-gnu/libnvinfer.so.8 (0x00007efe8e24d000)
    libspconv.so => /host_data/work2023/code/Lidar_AI_Solution/CUDA-BEVFusion/../libraries/3DSparseConvolution/libspconv/lib/x86_64/libspconv.so (0x00007efe8df63000)
    libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007efe8dd81000)
    libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007efe8dc32000)
    libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007efe8dc17000)
    libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007efe8da25000)
    /lib64/ld-linux-x86-64.so.2 (0x00007efe9cb92000)
    libcudart.so.11.0 => /usr/local/cuda/targets/x86_64-linux/lib/libcudart.so.11.0 (0x00007efe8d78c000)
    libprotobuf.so.17 => /lib/x86_64-linux-gnu/libprotobuf.so.17 (0x00007efe8d491000)
    libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007efe8d475000)

ll /usr/local/cuda/targets/x86_64-linux/lib/libcudart.so.11.0
    lrwxrwxrwx 1 root root 21 May  4  2021 /usr/local/cuda/targets/x86_64-linux/lib/libcudart.so.11.0 -> libcudart.so.11.3.109

ll /lib/x86_64-linux-gnu/libnvinfer.so.8
    lrwxrwxrwx 1 root root 19 Apr 27 06:21 /lib/x86_64-linux-gnu/libnvinfer.so.8 -> libnvinfer.so.8.6.1

ll /lib/x86_64-linux-gnu/libprotobuf.so.17
    lrwxrwxrwx 1 root root 21 Mar  9 17:15 /lib/x86_64-linux-gnu/libprotobuf.so.17 -> libprotobuf.so.17.0.0

```

Operations:
```
# 1.Update tensorrt
dpkg -i nv-tensorrt-local-repo-ubuntu2004-8.6.1-cuda-11.8_1.0-1_amd64.deb
sudo cp /var/nv-tensorrt-local-repo-ubuntu2004-8.6.1-cuda-11.8/nv-tensorrt-local-D7BB1B18-keyring.gpg /usr/share/keyrings/
apt-get update
apt-get install tensorrt

# 2.Update cudnn
dpkg -i cudnn-local-repo-ubuntu2004-8.9.0.131_1.0-1_amd64.deb
cp /var/cudnn-local-repo-ubuntu2004-8.9.0.131/cudnn-local-54F08CCB-keyring.gpg /usr/share/keyrings
apt-get update
apt-get install libcudnn8-dev

# 3.Install protobuf & onnx
25  apt install libprotobuf-dev
26  pip install onnx

# 4.Edit env variables
vim ./tool/environment.sh

# 5.Build & run
bash ./tool/environment.sh 
bash ./tool/build_trt_engine.sh
bash ./tool/run.sh
```

hopef, 2023-06-20:
I think you can print the version of TensorRT in the application to confirm if you are using a lower version.
From this information I observed, I think it may be caused by a lower version of TensorRT being used.

insmod-he, 2023-06-21:
I completely deleted the old version tensorrt, and then installed tensorrt 8.5. Now the detection results are correct.

closed 2023-06-22 by hopef

WangHeng1021, 2023-07-11:
Hello, I've got the same problem while I use cuda11.4 cudnn8.6.0 tensorRT8.6 and protobuf3.6.1.
Should I delete tensorRT8.6 and use tensorRT8.5?

insmod-he, 2023-07-11:
> Hello, I've got the same problem while I use cuda11.4 cudnn8.6.0 tensorRT8.6 and protobuf3.6.1. Should I delete tensorRT8.6 and use tensorRT8.5?

I think you need to make sure that you only have one version of TensorRT installed on your system. Otherwise, there may be conflicts.