NVIDIA-AI-IOT/Lidar_AI_Solution #99: Can you provide demo about onnx infer by TRT

lix19937, 2023-06-20:
https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/CUDA-CenterPoint/qat

By `python onnx_export/export-scn.py  --save-onnx=qat.onnx --use-quantization   --ckpt=onnx_export/QAT.pth`  can get qat.onnx, but I want to infer by tensorrt with qat.onnx,   Can you provide some demo about onnx infer by TRT  ?   Thanks.

hopef, 2023-06-20:
A good case is provided [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/master/CUDA-CenterPoint/src/centerpoint.cpp), right do you think?

closed 2023-06-21 by hopef

lix19937, 2023-06-21:
Thanks for your reply.   @hopef 

BTW, I want to `confirm` 5 understandings:     
1, CUDA-CenterPoint/src/centerpoint.cpp infer a model which include (backone + head) ?      
* pre_ (preprocess to net in)  
* scn_engine_ ( **backone**)   
* trt_ ( **head**)  
* post_  (postprocess from net out)      

2, For **centerpoint.scn.onnx**, which include some SparseConvolution(from `traveller59/spconv`), 
and **libspconv.so** just map to `centerpoint.scn.onnx` code implementation ?     

3, If my backbone is not the same as `scn`, can **libspconv.so** achieve one `SparseConvolution` op function ?     

4, In QAT phase, **SparseConvolution** as an independent layer, insert Q-DQ before SparseConvolution layer, 
that is `SparseConvolutionQuant`, For `scn.qat.onnx` from `export-scn.py`, **input_dynamic_range** map to Q-amax(per tensor quant),  **weight_dynamic_ranges** map to Q-amax(per channel quant) ?     

5, From `traveller59/spconv`, SparseConvolution include many steps, I want to know if it has been quantified internally in 
NV closed-source **libspconv.so** , not just layer IO Q-DQ ?

lix19937, 2023-06-22:
@hopef   Hi, hopef can you give me some advice about  https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/issues/99#issuecomment-1600101988  ?

mx2013713828, 2024-07-01:
> @hopef Hi, hopef can you give me some advice about [#99 (comment)](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/issues/99#issuecomment-1600101988) ?

Hello, did you solve your problem?I have same questions.