NVIDIA-AI-IOT/Lidar_AI_Solution #100: Does li'b's'p'co'nü support 3D

CreepingBird, 2023-06-21:


CreepingBird, 2023-06-21:
Hello, may I ask if there is any support for SparseInverseConv3d. I look forward to your reply.

lix19937, 2023-06-21:
I think, libspconv.so just map to `centerpoint.scn.onnx` code implementation.  
You can try it by setting `inverse=1` in SparseConvolution onnx node ATTRIBUTES.

hopef, 2023-06-22:
Sorry, currently there is no support for SparseInverseConv3d

CreepingBird, 2023-06-25:
> You can try it by setting `inverse=1` in **SparseConvolution** onnx node ATTRIBUTES and check the result. Maybe not support now.

Thank you for your reply. I still have some questions. What should I do during training, to remove the SparseInverseConv3d?

hopef, 2023-06-25:
I have no better recommendation to remove it.

CreepingBird, 2023-06-28:
> I have no better recommendation to remove it.

can F.interpolate be used instead of SparseInverseConv3d in backbone？

CreepingBird, 2023-06-28:
>

closed 2023-06-28 by CreepingBird

qqcx, 2024-02-20:
What a pity.