NVIDIA-AI-IOT/Lidar_AI_Solution #102: Image Data Normalization

alre5639, 2023-06-21:
Reading the "Image.tensor" data it looks like the image pixels are normalized between -1.7 and 1.5. Can you please provide some context on this normalization step?

hopef, 2023-06-22:
You can get the normalization steps by debugging the pytorch code of bevfusion.
Also you can see the cuda implementation of the code with the steps [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/3a45a0fc1283f50772d5ea472f4421e9564a8cd4/CUDA-BEVFusion/src/bevfusion/camera-normalization.hpp#L73).

hopef, 2023-06-22:
lidar2camera(3D -> 3D) is a conversion matrix based on lidar coordinates to camera coordinates.
    -  This conversion is still in physical space.
    
lidar2image(3D -> 2D) is a conversion matrix based on lidar coordinates to image coordinates. It is equivalent to a combine lidar2camera and camera2image.
    -  lidar2image = camera2image @ lidar2camera @ lidar_points
    -  This conversion converts physical space to image space (in pixel units)

alre5639, 2023-06-23:
sounds great thank you

closed 2023-06-23 by alre5639