NVIDIA-AI-IOT/Lidar_AI_Solution #103: Segmentation fault [Cuda-CenterPoint]

YoushaaMurhij, 2023-06-22:
I tried to rewrite loading data so it takes pointclouds from ROS Bag instead of files.
```cpp
        std::vector<float> points;
        points.reserve(width * height * fields.size());
        for (size_t i = 0; i < height; ++i)
        {
            for (size_t j = 0; j < width; ++j)
            {
                for (size_t k = 0; k < params.feature_num; ++k)
                {
                    for (const auto &field : fields)
                    {
                        size_t offset = field.offset + j * point_step + i * row_step + k * sizeof(float);
                        float value;
                        memcpy(&value, data.data() + offset, sizeof(float));
                        points.push_back(value);
                    }
                }
            }
        }

        unsigned int length = points.size();
        
        size_t points_num = length / (fields.size() * 5);
        void* pc_data = new float[points.size()];
        std::memcpy(pc_data, points.data(), points.size() * sizeof(float));

        std::cout << "find points num: " << points_num << std::endl;

        checkCudaErrors(cudaMemcpy(d_points, buffer, length, cudaMemcpyHostToDevice));

        std::cout<<"start infer" << std::endl;
        centerpoint.doinfer((void *)d_points, points_num, stream);
```
but, I got segmentation fault error: in `centerpoint.doinfer((void *)d_points, points_num, stream);`
      
 ```bash
       GPU has cuda devices: 1
----device id: 0 info----
  GPU : NVIDIA GeForce RTX 3060 
  Capbility: 8.6
  Global memory: 12041MB
  Const memory: 64KB
  SM in a block: 48KB
  warp size: 32
  threads in a block: 1024
  block dim: (1024,1024,64)
  grid dim: (2147483647,65535,65535)

find points num: 131072
start infer
[ros2run]: Segmentation fault
 ```
 Could you please suggest me what could be the problem?
 It works fine in case of loading data from files.
Thanks!

lix19937, 2023-06-23:
Maybe `illegal address`,   you can `set(CMAKE_BUILD_TYPE "Debug")` in CMakeLists.txt, and then use gdb to debug.

YoushaaMurhij, 2023-06-23:
I think so. I have set debug mode.
Can you please coorect me if I am wrong?
- d_points allocation size should be >= pc_data size
- length = num of bytes in pc_data and could <= num of bytes in d_points

Thanks

lix19937, 2023-06-23:
Yes,  you can use follow  define `d_points` :    
```
    std::cout << "get length/sizeof: " << msg->width * msg->height  * params.feature_num << std::endl;
    float *d_points = nullptr;    
    checkCudaErrors(cudaMalloc((void **)&d_points, msg->width * msg->height  * params.feature_num * sizeof(float)));  
```

then **change** from follow  
```
        float* points = new float[length];
        for (size_t i = 0; i < points_num; ++i)
        {
            const uint8_t* data = &msg->data[i * msg->point_step];

            float x = *(float*)(&data[0]);
            float y = *(float*)(&data[4]);
            float z = *(float*)(&data[8]);
            float n = *(float*)(&data[12]); // intensity
            float t = *(float*)(&data[16]); // time

            points[i * params.feature_num] = x;         
            points[i * params.feature_num + 1] = y;
            points[i * params.feature_num + 2] = z;
            points[i * params.feature_num + 3] = n;
            points[i * params.feature_num + 4] = t;
        }

```
 to 

```
float* points = new float[msg->width * msg->height * params.feature_num ];
```
@YoushaaMurhij

YoushaaMurhij, 2023-06-23:
Thanks for your response. I tried that, everything seems to be correct and the log, too. But still not working 
before that,  I used:
`checkCudaErrors(cudaMalloc((void **)&d_points, MAX_POINTS_NUM * params.feature_num * sizeof(float)));`

YoushaaMurhij, 2023-06-23:
Debug shows that the seg. fault happens [here](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/3a45a0fc1283f50772d5ea472f4421e9564a8cd4/CUDA-CenterPoint/src/preprocess_kernels.cu#L76). I am pretty sure the input format is wrong, but can't find the problem.

YoushaaMurhij, 2023-06-23:
I think that libspconv.so is not compatible. I am running it on ubuntu 22.04 + cuda 12.1  +  c++21

closed 2023-06-25 by YoushaaMurhij