NVIDIA-AI-IOT/Lidar_AI_Solution #105: Cannot create TRT engine from centerpoint.scn.onnx

akjt, 2023-06-25:
Hi, 
I am trying to build the tensorRT engine from the centerpoint.scn.onnx file. But I get following error:

3: getPluginCreator could not find plugin: SparseConvolution version: 1
[06/25/2023-11:55:45] [E] [TRT] ModelImporter.cpp:771: While parsing node number 0 [SparseConvolution -> "1"]:
[06/25/2023-11:55:45] [E] [TRT] ModelImporter.cpp:772: --- Begin node ---
[06/25/2023-11:55:45] [E] [TRT] ModelImporter.cpp:773: input: "0"

Do you have any ideas how I can create the tensorRT engine from the above onnx file.

hopef, 2023-06-25:
This ONNX file can only be run by [libspconv.so](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/tree/master/libraries/3DSparseConvolution).

akjt, 2023-06-25:
Is the source code to libspconv.so available? I noticed that this library is not the same as this [spconv](https://github.com/traveller59/spconv).

Also, it says that libspconv only support sm > 80 - so does this mean that I cannot run the trtengine on architectures like Xavier (which supports up to sm 75

akjt, 2023-06-25:
@hopef Is the code behind libspconv.so available open source - or will you in the near future support sm 75?

akjt, 2023-06-29:
@hopef Do you have an update on my questions, please?

hopef, 2023-07-06:
Sorry, there are no plans to support devices with sm<80.

hopef, 2023-07-06:
For updates to libspconv.so, you can submit your comments. 
But for libspconv open source, we need to wait for a signal from the leader.

qqcx, 2023-07-21:
This will help you. SubMConv3d, set : bias=True

HenryZhangJianhe, 2023-09-20:
> This will help you. SubMConv3d, set : bias=True

Did you test libsparse.so passed on the xavier?

qqcx, 2023-09-22:
> > This will help you. SubMConv3d, set : bias=True
> 
> Did you test libsparse.so passed on the xavier?

Never before.