NVIDIA-AI-IOT/Lidar_AI_Solution #109: result is incorrect

MarioUncle-Wang, 2023-06-27:
Hello, 
          My environment is configured as shown in the figure. As in question #97, my TensorRT version should be correct. But I have the same problem with him, what is the reason?
          Besides, running the program multiple times, the outputs are different each time.
[image]

closed 2023-06-29 by MarioUncle-Wang

zhangshen12356, 2024-07-01:
> Hello, My environment is configured as shown in the figure. As in question #97, my TensorRT version should be correct. But I have the same problem with him, what is the reason? Besides, running the program multiple times, the outputs are different each time. ![kkk](https://private-user-images.githubusercontent.com/81735762/249021593-698ee1e7-486b-43b0-8b8f-af8942a9c1a8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk4MDU3NzcsIm5iZiI6MTcxOTgwNTQ3NywicGF0aCI6Ii84MTczNTc2Mi8yNDkwMjE1OTMtNjk4ZWUxZTctNDg2Yi00M2IwLThiOGYtYWY4OTQyYTljMWE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzAxVDAzNDQzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlhODcxZWI4YTMwNjFhZGZlODU2NjBmZWViNjY1NWQ0YjlkNmVmNWVmYjU2MTM4NTQ3YzhlMmQ2OWY0YjJlM2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.QC4IpYFMCUkNEOLIZ6Hme_LnmphBBKWS5cBZ_KZQBgU)

how did you solve it?

MarioUncle-Wang, 2024-07-04:
> how did you solve it?

Most likely due to a conflict in the TRT version. You need to make sure that you only have one version of TensorRT installed on your system.