NVIDIA-AI-IOT/Lidar_AI_Solution #110: realtime detection: img_aug_matrix update

SeungHwi0613, 2023-06-27:
Hello! first, i really appreciate about this wonderful work.

I try to use this for realtime bevfusion detection, that's why i have to change every static dataset to sensor data. 

But i have one question about your code.

in main.cpp line318

why we need img_aug_matrix for updating?
i though this code is generating result image of detection and updating core for this. 
as far as i know, augmentation is for training. 

If you give me some advise to understand the meaning of 'img_aug_matrix' and reason of this code, it could be so nice!

Thanks.

hopef, 2023-07-03:
The meaning of img_aug_matrix is to record [how to resize, crop, and rotate the image](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/pipelines/transforms_3d.py).

[The resize, crop, and rotate used are random for the training phase](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/pipelines/transforms_3d.py#L41C20-L41C20).
For the inference phase, [the resize, crop, and rotate used is fixed(no rotation when inference)](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/pipelines/transforms_3d.py#L52C19-L52C19), e.g., [resize_lim(0.48)](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/configs/nuscenes/default.yaml#L215C19-L215C28).

Since BEVFusion uses lidar points to [enhance the camera's feature extraction](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/vtransforms/depth_lss.py).
[To align the lidar with the camera](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/vtransforms/base.py#L262C34-L262C34). It needs to know how the camera is cropped and scaled for alignment.

So img_aug_matrix is a training-independent matrix.

SeungHwi0613, 2023-07-03:
> The meaning of img_aug_matrix is to record [how to resize, crop, and rotate the image](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/pipelines/transforms_3d.py#L26C2-L26C2).
> 
> [The resize, crop, and rotate used are random for the training phase](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/pipelines/transforms_3d.py#L41C20-L41C20). For the inference phase, [the resize, crop, and rotate used is fixed(no rotation when inference)](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/datasets/pipelines/transforms_3d.py#L52C19-L52C19), e.g., [resize_lim(0.48)](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/configs/nuscenes/default.yaml#L215C19-L215C28).
> 
> Since BEVFusion uses lidar points to [enhance the camera's feature extraction](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/vtransforms/depth_lss.py). [To align the lidar with the camera](https://github.com/mit-han-lab/bevfusion/blob/db75150717a9462cb60241e36ba28d65f6908607/mmdet3d/models/vtransforms/base.py#L262C34-L262C34), It needs to know how the camera is cropped and scaled.
> 
> So img_aug_matrix is a training-independent matrix.
> 
> Of course, you can still pass in only a scale value and two crop values(x,y) when computing the camera-depth, instead of a matrix called img_aug_matrix.

Thank you for kind answer!! It is so helpful to me. 
When i checked data["img_aug_matrix" it was just eye(4) matrix. Is it normal?

hopef, 2023-07-03:
For the nuScenes dataset(1600x900), when the model uses a resolution of 704x256. It is:
```bash
  [[ 0.48 0. 0. -32. ]
   [ 0. 0.48 0. -176. ]
   [ 0. 0. 1. 0. ]
   [ 0. 0. 0. 1.]]
```

For your model, it may be an identity matrix if you set the corresponding parameters.

Otherwise, you may encounter an error.
As I said, this matrix stores the image's scale and crop parameters. That is a general operation on nuScenes.

SeungHwi0613, 2023-07-06:
> For the nuScenes dataset(1600x900), when the model uses a resolution of 704x256. It is:
> 
> ```shell
>   [[ 0.48 0. 0. -32. ]
>    [ 0. 0.48 0. -176. ]
>    [ 0. 0. 1. 0. ]
>    [ 0. 0. 0. 1.]]
> ```
> 
> For your model, it may be an identity matrix if you set the corresponding parameters.
> 
> Otherwise, you may encounter an error. As I said, this matrix stores the image's scale and crop parameters. That is a general operation on nuScenes.

Thank you i really appreciate about it.

closed 2023-07-06 by SeungHwi0613

alre5639, 2023-07-24:
In regards to the img_aug_matrix I would like to modify it to use images that are cropped 88 pixels on top and bottom rather that 176 on top (our cameras are pointed down so we lose alot of information cropping the top of the image). would I just use the following matrix?:
```
[[ 0.48 0. 0. -32. ]
   [ 0. 0.48 0. -88. ]
   [ 0. 0. 1. 0. ]
   [ 0. 0. 0. 1.]]
```
using this matrix with the provided pretrained network should work fine right?