NVIDIA-AI-IOT/Lidar_AI_Solution #111: The result of int8 inference is incorrect

Oh-Cest-la-vie, 2023-06-27:
Hello 

When I infence(CUDA) bevfusion Int8, I find a problem in both A4000 and Drive Orin. The result of fp16 is correct, but the inference with Int8 losed most targets.

[fp16]
[image]

[Int8]
[image]

I use TensorRT 8.4.1.5 in A4000 and DRIVE OS 6.0 TensorRT 8.4.11 in Drive Orin.

Oh-Cest-la-vie, 2023-06-27:
I use TensorRT 8.5.3 in A4000, the inference result is correct now. But the latest version of TensorRT in Drive Orin is TensorRT 8.4.12, how can I solve this problem? If you have any other advice, besides upgrading the version of TensorRT?

lix19937, 2023-06-29:
@Oh-Cest-la-vie  you can update DRIVE OS 6.0.6.0, which TensorRT is v8510

Oh-Cest-la-vie, 2023-06-30:
Got it, thx @lix19937

closed 2023-07-06 by Oh-Cest-la-vie

rayxie11, 2024-05-14:
Hi @Oh-Cest-la-vie, I am also using Drive Orin platform. I was not able to get the correct result in neither int8 nor fp16. I was wondering if you have solved this issue and perhaps provide me with some advice on how to make it work on Drive Orin. My email address is rayxie11@gmail.com. I'm also fine with leaving messages here. Thank you!