NVIDIA-AI-IOT/Lidar_AI_Solution #114: Spconv support gpu arch

xiaoxifuhongse, 2023-06-29:
gpu : NVIDIA GeForce RTX 2080 SUPER 
system: ubuntu20.04
cuda: 11.7

when i run Lidar_AI_Solution-master/Libraries/3DSparseConvolution/tool/pytest.py then

yunqiushuiman, 2023-07-20:
Thank you for the expert's answer to this question. I am facing the same issue as well. CUDA Runtime error init_quantization_scale<FeaturesType::INT8><<<grid, block, 0, _stream>>>( input_scale, output_scale, weight_, prod_DHWI, int8weight_.ptr(), weight_scales_.ptr(), alpha_.ptr(), (half *)bias, bias_scaled_.ptr(), numel) # no kernel image is available for execution on the device, code = cudaErrorNoKernelImageForDevice [ 209 ] in file src/spconv/implicit-gemm.cu:241
tool/run.sh: line 41: 11487 Aborted                 (core dumped) ./build/bevfusion $DEBUG_DATA $DEBUG_MODEL $DEBUG_PRECISIONIf I want to run this program on a low-powered laptop, what can I do to make it work? How can I modify the program to make it suitable for a computing power of 75?

yunqiushuiman, 2023-07-20:
I want to run this program on a low-powered laptop, what can I do to make it work? How can I modify the program to make it suitable for a computing power of 75?

hopef, 2023-07-20:
Sorry. Currently, the libspconv.so can not run on sm75 or lower.