NVIDIA-AI-IOT/Lidar_AI_Solution #116: error occurred in run tool/dump-data.py

JQZhai, 2023-07-05:
I am facing a problem like this
```
run python tool/dump-data.py
ImportError: cannot import name 'build_dataloader' from 'mmdet3d.datasets'
```
how to solve it, thanks

bigsheep2018, 2023-07-06:
@JQZhai Obviously you need bevfusion codebase/environment for the missing modules.
Set export path in your terminal and check the bevfusion folder is fully cloned.

JQZhai, 2023-07-06:
> @JQZhai Obviously you need bevfusion codebase/environment for the missing modules. Set export path in your terminal and check the bevfusion folder is fully cloned.

Thank you for your reply. After checking the source code, I found that mmdet3d has undergone major changes after v1.0.0. For example, the module "mmdet3d.datasets" no longer exists. So the ”tool/dump-data.py“ file may need to be modified according to the new API.

bigsheep2018, 2023-07-06:
@JQZhai 
Just checked.
The default version of mmdet3d codebase provided by this repo's Readme (git clone --recursive) works fine. 
Yes, for the newer releases of mmdet3d repo, you need to modify the api accordingly. good luck.

JQZhai, 2023-07-06:
@bigsheep2018 
Sadly, I downloaded the repo using the git clone --recursive command and added the mmdet3d directory in bevfusion (@db75170) to tool/dump-data.py, and the following error occurred: 
`ImportError: cannot import name 'ball_query_ext' from partially initialized module 'mmdet3d.ops.ball_query`'.
How should I fix this error, thanks a lot

hopef, 2023-07-06:
You should refer to the installation tutorial in the bevfusion repo.

hopef, 2023-07-13:
Using an old commit of BEVFusion via the command below will fix this issue:

```bash
git checkout db75150717a9462cb60241e36ba28d65f6908607
or
git reset --hard db75150717a9462cb60241e36ba28d65f6908607
```

closed 2023-07-18 by hopef