NVIDIA-AI-IOT/Lidar_AI_Solution #117: how to get/train the pretrained resnet50xxx.pth and swin_xxx.pth

bigsheep2018, 2023-07-05:
Hello @hopef 
I found a issue mentioned the problem on official bevfusion repo. Link: https://github.com/mit-han-lab/bevfusion/issues/176
It shows a quite complicated way for getting the pre-trained swin_xxx.pth. 
I.e., we need to train a CascadeRCNN on Nuscenes 2D detection task following a pretrained Swin-T on COCO 2D detection task. 
Well, as we need to train bevfusion from scratch on customized dataset (2D det pretrain --> 3D det bevfusion), there are 2 questions need to be checked:
1. Is there an easier way to get the pretrained swin_xxx.pth and resnet50xxx.pth? Or maybe is it possible to release the training config/code of these pretrained .pths?
2. For using Resnet50xxx.pth, are you guys just skipping the pre-training procedures on COCO/nuScenes 2D Det task and using the default pytorch pretrained model directly?

hopef, 2023-07-05:
https://github.com/open-mmlab/mmdetection3d/tree/1.0/configs/nuimages

bigsheep2018, 2023-07-06:
> https://github.com/open-mmlab/mmdetection3d/tree/1.0/configs/nuimages

Thanks for the reply @hopef . I would try Cascade MaskRCNN and HTC's backbones, then get back to this issue.

bigsheep2018, 2023-07-10:
> https://github.com/open-mmlab/mmdetection3d/tree/1.0/configs/nuimages

Hello @hopef ,
Using HTC's  pretrained resnet50 backbone, 
we got **MAP 67.98** and **NDS 71.03** on Nuscenes Validation set. 
Closing the issue.

closed 2023-07-10 by bigsheep2018