NVIDIA-AI-IOT/Lidar_AI_Solution #119: Does 3D backbone support PTQ

akjt, 2023-07-06:
I see the example regarding centerpoint scn 3D backbone which includes fusion and QAT. 

Does this library support the use of PTQ instead - and in case where can I look for an example?

superpigforever, 2023-07-06:
https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/3a45a0fc1283f50772d5ea472f4421e9564a8cd4/CUDA-CenterPoint/qat/tools/centerpoint_qat_pipline.py#L216

superpigforever, 2023-07-06:
https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/blob/3a45a0fc1283f50772d5ea472f4421e9564a8cd4/CUDA-CenterPoint/qat/tools/centerpoint_qat_pipline.py#L216

Set ptq_mode in the args to True

akjt, 2023-07-06:
But is that not only for training? 
I already have a trained model, and wanted to use the fusion+quantisation - so i am using this  `centerpoint_eval.py` , however, the problem is that it wants to fuse before loading the weights, which won't work for PTQ scenario since you dont compress the weights

hopef, 2023-07-07:
PTQ is the first step of QAT. You can modify it yourself to achieve your purpose.
We need to sort out the relationship between PTQ and layer-fuse and handle it properly, as done in the code.