NVIDIA-AI-IOT/Lidar_AI_Solution #121: Support of PACT

Oh-Cest-la-vie, 2023-07-11:
Hello,

Thanks for your work first, I'm trying PACT quantization and I add a [clip function](https://github.com/NVIDIA/TensorRT/blob/a167852705d74bcc619d8fad0af4b9e4d84472fc/tools/pytorch-quantization/pytorch_quantization/nn/modules/clip.py) equipped with backward (which I find in TensorRT pytorch quantization) in front of those ReLU in the Spconv Encoder. I want to know if libspconv.so also support deployment a PACT module like that ?

hopef, 2023-07-12:
Currently, the libspconv doesn't implement the PACT quantization. I'm interested in PACT, and we'll document it as a basis for future improvements.

Oh-Cest-la-vie, 2023-07-14:
Got it, thanks for your reply.

closed 2023-07-14 by Oh-Cest-la-vie