NVIDIA-AI-IOT/Lidar_AI_Solution #123: SparseConv2d and SubMConv2d support

HuangVictorAuto, 2023-07-12:
Hi, thanks for the great work.

I have used some spconv 2d layers in my network. I read your code, I see that the 3D versions are all converted to 'SparseConvolution' onnx node and will be parsed and used in spconv engine during inference. I wonder if the engine or libspconv also compatialbe with the sparse 2D versions? I can also convert the 2d versions to 'SparseConvolution' onnx node?

thanks!

hopef, 2023-07-12:
The 2d SPCONV is a specific form of 3d SPCONV. You just set z to 1 at indices which can achieve the 2d SPCONV.

HuangVictorAuto, 2023-07-13:
Thanks for the reply. I will covert some parameters of SPCONV2D to suit the 'SparseConvlution' parameters.

AhmedNasr7, 2023-09-12:
@HuangVictorAuto  Hi! can you explain how did you convert it if you already did? thanks!

HuangVictorAuto, 2023-09-12:
@AhmedNasr7 , just as above mentioned, 2d sparsconv is just a special case for 3d sparseconv, just set z dimension to 1. 
you can update your pytorch network from 2d to 3d first, and then do the onnx export.

AhmedNasr7, 2023-09-12:
I guess this requires re-training the network, no out of the box solutions for pretrained models?

HuangVictorAuto, 2023-09-13:
I manage to load a pretrained model, update the 2d spconvs to 3d spconv weights and bias accordingly to suit the libspconv 'sparseconvlution'.

HuangVictorAuto, 2023-09-14:
@hopef , 
I manage to convert the 2d sparseconv/submanifold to 3d, by two methods, update the weights and shapes during the onnx node create stage. And I also update the pytorch model from 2d to 3d and then covert to onnx.

But both method, I found that the result of sparseconv2d is aligned between pytorch and engine. But the result from submanifold 2d is not aligned. Have you test the libspconv  submanifold3d (with special case z=1 ) situation?

hopef, 2023-10-19:
@HuangVictorAuto 
could you give me a reproduce data?
For example a simple pytorch code and a simple input data(features, indices, grid_size).

HuangVictorAuto, 2023-10-25:
Hi, attached file is a short sample for subm2d network. please check it. 
I found big difference between the pytorch and engine result, just for 1 subm2d node.
[subm2d.zip](https://github.com/NVIDIA-AI-IOT/Lidar_AI_Solution/files/13162313/subm2d.zip)