NVIDIA-AI-IOT/Lidar_AI_Solution #124: points.tensor float16

MarioUncle-Wang, 2023-07-12:
Dear author, 
    The lidar pointcloud saved in "points.tensor" is float16 data type. I want to process the pointcloud data directly in the C++ program, but C++ only has float32 data type. How to converte from float32 to float 16, or is it possible to not do the conversion and directly use float32?

hopef, 2023-07-12:
There have three solutions below:
1. Do enforce data type conversion in the CUDA kernel.
2. Using the CUDA API `__float2half`.
3. Using the C++ conversion code directly.

```c++
typedef unsigned short half;
static inline half __internal_float2half(const float f)
{
    unsigned int x;
    unsigned int u;
    unsigned int result;
    (void)memcpy(&x, &f, sizeof(f));
    u = (x & 0x7fffffffU);
    sign = ((x >> 16U) & 0x8000U);
    // NaN/+Inf/-Inf
    if (u >= 0x7f800000U) {
        remainder = 0U;
        result = ((u == 0x7f800000U) ? (sign | 0x7c00U) : 0x7fffU);
    } else if (u > 0x477fefffU) { // Overflows
        remainder = 0x80000000U;
        result = (sign | 0x7bffU);
    } else if (u >= 0x38800000U) { // Normal numbers
        remainder = u << 19U;
        u -= 0x38000000U;
        result = (sign | (u >> 13U));
    } else if (u < 0x33000001U) { // +0/-0
        remainder = u;
        result = sign;
    } else { // Denormal numbers
        const unsigned int exponent = u >> 23U;
        const unsigned int shift = 0x7eU - exponent;
        unsigned int mantissa = (u & 0x7fffffU);
        mantissa |= 0x800000U;
        remainder = mantissa << (32U - shift);
        result = (sign | (mantissa >> shift));
        result &= 0x0000FFFFU;
    }
    unsigned short x = static_cast<unsigned short>(result);
    if ((remainder > 0x80000000U) || ((remainder == 0x80000000U) && ((x & 0x1U) != 0U))) {
        x++;
    }
    return x;
}
```

MarioUncle-Wang, 2023-07-13:
Thank you for your reply!

closed 2023-07-13 by MarioUncle-Wang