NVIDIA-AI-IOT/Lidar_AI_Solution #125: [Feature Request] special operations for VoxelNext model

HuangVictorAuto, 2023-07-12:
Hi, 

we are trying to deploy the voxelnext model based on your libspconv. But voxelnext model have some special operations as follows, It is not supported by your current libspconv, can you add these to your ops?

1. fuse sparse features from different scales. we may need sparse concat feature for this.
```
x_conv5.indices[:, 1:] *= 2
x_conv6.indices[:, 1:] *= 4
x_conv4 = x_conv4.replace_feature(torch.cat([x_conv4.features, x_conv5.features, x_conv6.features]))
x_conv4.indices = torch.cat([x_conv4.indices, x_conv5.indices, x_conv6.indices])
```

2. bevout, from 3d directly to 2d, negelate the z dimension
```
def bev_out(self, x_conv):
        features_cat = x_conv.features
        indices_cat = x_conv.indices[:, [0, 2, 3]]
        spatial_shape = x_conv.spatial_shape[1:]

        indices_unique, _inv = torch.unique(indices_cat, dim=0, return_inverse=True)
        features_unique = features_cat.new_zeros((indices_unique.shape[0], features_cat.shape[1]))
        features_unique.index_add_(0, _inv, features_cat)

        x_out = spconv.SparseConvTensor(
            features=features_unique,
            indices=indices_unique,
            spatial_shape=spatial_shape,
            batch_size=x_conv.batch_size
        )
        return x_out
```

thanks!

hopef, 2023-07-12:
I think you can split it into multiple onnx files. Implement what you need by combining them.

HuangVictorAuto, 2023-07-13:
Thanks for your reply. I've thought that as a backup plan.

closed 2023-07-13 by HuangVictorAuto